openscopeproject/InteractiveHtmlBom #358: SHEETPATH variable in KiCad V7 not correct across hierarchies

myamigo, 2023-02-17:
I want using this plugin to aggregate capacitance and inductance within each hierarchal block and figured I could use add the ${SHEETPATH} variable, which is now supported in KiCad V7, to the symbol as follows:

[image]

The field is properly evaluated in the 3 hierarchical sheets where it is used... 
[image]

I intend to parse the SHEETPATH output to identify the root sheet value (data_path#) for all components nested under that sheet.

With the field added to the symbols I would then add the new symbol field name to the **ibom.config.ini** file. This goes in the [fields] section as follows...

```
[fields]
show_fields=Value,Footprint,Sheet_Path
group_fields=Value,Footprint,Sheet_Path
 ```

I selected the option to group by the new field as well so this should give me three lines for each common part but it does not... 

[image]

So I changed the settings to produce an ungrouped BOM... 

[image]

... to see what the script found...

[image]

... but rather than seeing 3 lines for the part with each indicating a different _**data_path#**_ root, I see that they are all the same (all based on data_path0). The 2nd & 3rd lines should indicate data_path1 & data_path2 respectively. 

We know that these fields are rendered correctly in KiCad so this must be an issue with the plugin.

qu1ck, 2023-02-17:
What are you using as extra data source?
[image]

If it's the pcb file then you need to pull changes from schematic before generating ibom.
[image]

If it's a netlist file then you need to regenerate netlist.

myamigo, 2023-02-17:
I am using the PCB and I am pulling from the schematic.

qu1ck, 2023-02-17:
Yeah, I just tested this and it's a bug in kicad. If you open the pcb file in text editor and search for "sheet_name" you will see it has the same value for all instances of components regardless which sheet they came from.

But the workaround is to use the built in sheetname field instead of adding your own. Does that work for you?

myamigo, 2023-02-17:
I'm trying to use ${SHEETPATH} rather than ${SHEETNAME}. I'll take a workaround  but I'm not clear on what you're suggesting.

myamigo, 2023-02-17:
I do see real issues in the PCB file.  Initially I had just 1 part with the Sheet_Path = **SHEETPATH** but I just added another. Now I am seeing the evaluated **SHEETPATH** value from the new part, which is on a different path, showing up in both places. I think I can generate an issue  in KiCad for this.

I understand now that your script uses  the already evaluated contents of the field. I though it was evaluating the SHEETPATH variable on the fly.

Thanks for the quick feedback,

qu1ck, 2023-02-17:
Pcbnew generates sheetname and few other fields automatically
[image]

I thought you could use that instead of path but because in your case the different part of path is the portion in the middle and not the end, it won't work.

myamigo, 2023-02-17:
All good! I know what to do to get this raised as an issue.

qu1ck, 2023-02-17:
https://gitlab.com/kicad/code/kicad/-/issues/13980

myamigo, 2023-02-17:
Very good. I'll add my test file to that. It includes the SHEETNAME and SHEETPATH variables in the symbol so you can see the disconnect when used in the PCB.

myamigo, 2023-02-17:
I do have a solution though. The SHEETPATH variable is supported in the Schematic Editor and the KiBOM python script handles the SHEETPATH variable the way I want so I'll use that for my data analysis and your plugin to make it easy to find parts, which it does very well.

qu1ck, 2023-05-31:
This is fixed in kicad 7.0.5

closed 2023-05-31 by qu1ck