openscopeproject/InteractiveHtmlBom #361: Add support for Kicad's native DNP field

Shackmeister, 2023-03-01:
In Kicad 7 there is now a native/dedicated option for DNP.
Ultimately this could replace the whole dnp selection section.

Thank you for a great plugin!

iyzpx, 2023-04-04:
Created a pull request: #370

Currently it's just ignoring/hiding the components in the bom flagged with dnp. IMO this might work now, but I would imagine it is better to mark the components as the kicad schematic does:

[image]

qu1ck, 2023-04-04:
There is an option to draw dnp components in outline mode. Cross doesn't work as it does in schematic because on pcb components can overlap.

andrewgreenberg, 2023-08-18:
Just bumping this; it would be GREAT to support KiCAD 7 DNP!

qu1ck, 2023-08-18:
Native dnp field is now supported, will be in next version. To use it select `kicad_dnp` as DNP in fields tab in config

[image]

commit f918435b (2023-08-18): Add kicad 7 native dnp field support

Fixes #361
closed 2023-08-18 by qu1ck