openscopeproject/InteractiveHtmlBom #364: Dimensions not supported => crash

nopeppermint, 2023-03-22:
Thanks for your nice tool.

I found a bug today, in my PCB I added some dimensions and apparently your tool do not proceed them correctly.
[image]

It only happens if you place "dimensions" on the F.Silkscreen layer, if they are on a user layer it does not crash.

[image]

When I removed them from the PCB, there was no longer any crash.

Attached an example kicad 7 project where the bug is reproducible:
[dimensions_crash.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/11037875/dimensions_crash.zip)

Interactive BOM v2.5

My Kicad version:

```
Application: KiCad PCB Editor x64 on x64

Version: 7.0.1-41-g121ea14fe4, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.12.1
	HarfBuzz 6.0.0
	FontConfig 2.14.1
	libcurl/7.88.1-DEV Schannel zlib/1.2.13

Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Mar 20 2023 04:13:40
	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.81.0
	OCC: 7.6.3
	Curl: 7.88.1-DEV
	ngspice: 39
	Compiler: Visual C++ 1934 without C++ ABI

Build settings:
	KICAD_SPICE=ON

```

qu1ck, 2023-03-23:
Fix is now in v2.6

closed 2023-03-23 by qu1ck

nopeppermint, 2023-03-24:
thanks @qu1ck