openscopeproject/InteractiveHtmlBom #365: Big Wish for BOM

WisdomCounselling, 2023-03-25:
Hi. A big wish.  Can you add a column so that JLCPCB part numbers can be added in for parts?  Then a button to create a JLCPCB output spreadsheet?
That would I think make this already amazing plug-in perfect!
Thank you. Michael
[image]

qu1ck, 2023-03-25:
> Can you add a column so that JLCPCB part numbers can be added in for parts?

I can't but you can. There is a generic way to add any field you want, go to fields tab when generating bom, any field you add in schematic editor will show up there, just make sure to update the board.

> Then a button to create a JLCPCB output spreadsheet?

I don't know what "JLCPCB spreadsheet" means but there is also a button to copy the bom into clipboard, you can paste it into spreadsheet and modify as you wish.

WisdomCounselling, 2023-03-27:
Great, thanks so much!  I tried the value R1, 10k, and it adds the column for ALL 10k resistors - which is really helpful, I don't have to update all 10k resistors in the schematic editor. Like it.
The copy BOM feature is perfectly adequate...
Thanks again. M

WisdomCounselling, 2023-03-27:
Oh, OK it didn't work as a group second time around.

closed 2023-03-28 by qu1ck