openscopeproject/InteractiveHtmlBom #366: Eagle: the script doesn't generate output if the outline is drawn as a component

depalamarchuk, 2023-03-29:
The script doesn't see Dimension layer in library. It can be easily fixed by copying the outline from library to board though. The attachment has a minimum (not) working example.
[iBOM_dimension_test.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/11102058/iBOM_dimension_test.zip)

qu1ck, 2023-03-29:
Thanks for bug report with test project.

@Funkenjaeger wanna take a look?

Funkenjaeger, 2023-03-29:
Will do.

Funkenjaeger, 2023-03-29:
Fix pushed, PR #367.  I second what @qu1ck said above - an [SSCCE ](http://sscce.org/) is always appreciated, thank you!
[image]

closed 2023-03-30 by qu1ck

depalamarchuk, 2023-03-30:
Hello,

Thank you for fixing it fast. However, with some shapes it still doesn't work. Probably it cannot process arcs connected to arcs. Sometimes it does work, so I cannot reproduce the bug consistently. Deleting stuff at 103 Test layer (not present here) also helped. I'm not sure what is going on here. But copying it to board still works as a temporary fix. I'm attaching another example board.

[iBOM_dimension_test2.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/11108584/iBOM_dimension_test2.zip)

Funkenjaeger, 2023-03-31:
Sorry about that @depalamarchuk, I overlooked a one-line change in my previous fix.  I pushed a fix in my fork (97cae23) - to perhaps learn a lesson from my earlier haste, why don't you give it a try before I submit another PR :)

depalamarchuk, 2023-03-31:
@Funkenjaeger Thank you for a fix. Now it works with complex shapes.

qu1ck, 2023-04-01:
@Funkenjaeger please open a PR with that commit.

commit 1b551f52 (2023-03-30): (Eagle) Handle edges in footprints (fix #366)