openscopeproject/InteractiveHtmlBom #368: Using local config file with CLI mode

yagui, 2023-03-31:
I'm running the script from command line and I set the environment variable INTERACTIVE_HTML_BOM_NO_DISPLAY.
This works ok, but even though the ibom.config.ini file is present and also the general config.ini file is located at the InteractiveHtmlBom folder, neither of those is read in this NO_DISPLAY mode.
I there a way to force it to use either one of the config files?

I'm using latest (at the moment) commit 1b551f52dfc9e9347b48e1c9c7f731cef0f02562 and KiCad 7.0.1

Thanks

qu1ck, 2023-03-31:
No, currently in cli mode config files are not read, only flags you pass on command line matter.

wxWidgets has (or had) a bug where config reader does not work properly if gui is not initialized. I would probably need a hand-written parser to make config files work in cli mode.

yagui, 2023-03-31:
I there an example on how to pass those flags through the command line? I couldn't find them.

I was thinking to force a simple parser if the option of no display is detected in the environment variables. I'll try to add it and if it works I'll submit a pull request.

qu1ck, 2023-03-31:
If anyone wants to take a stab at it, be my guest. Only requirement is to not use any packages outside of standard python installation and to test your code at least on windows and linux. I can help test on windows if you can't.

> I there an example on how to pass those flags through the command line? I couldn't find them.

Launch the script with `--help` flag. Or read wiki usage page.

> why not using other mechanism for the config file?

wxWidgets config lib was convenient and there wasn't a good reason to switch. Like you said, for cli it's easy to write a wrapper script to set all the flags you want.

brockn8r, 2023-11-10:
qu1ck, I do have a good starting point for this, and I have created a patch if you are interested. I could create a branch and push, but I have not done any thorough testing. It works ok for what we need.

qu1ck, 2023-11-10:
You can open a draft pull request, I can at least leave some comments on whether you are going in a direction that can be merged and what work needs to be done, if any, for that.

hildogjr, 2025-01-07:
The local project configuration will allow a good integration with the new jobset output manager of KiCad. just running iBOM as as CLI command.