openscopeproject/InteractiveHtmlBom #371: Parts included in bill of materials and exluded from board are not shown in the interactive bom

bwack, 2023-04-07:
Lately I've figured out it is possible to add (digikey etc) order numbers to the ibom by using the symbols field table in the schematic editor, and when creating the ibom, I select that field in the InteractiveHtmlBom form, Fields tab:

[image]

The result is great:
[image]

Though, sometimes in large projects like mine, you have multiple orders per part. Say you want DIP sockets with your 74LS14 ICs, or you have this large glass fuse, but you also need the fuse holder. In my case that is the glass fuse + two fuse clips. My solution to this was to add symbols in the schematics indicating the fuse clips, and in their properites set them to be in the bom, but not on the board, only the fuse is on the board and linked to an actual footprint. So I wonder, why they don't show up in the ibom ?
Is it because the interactivehtmlbom script only looks in the pcbnew file, and not in the schematics file ?

qu1ck, 2023-04-08:
Yes, it's because ibom only uses things that are on the board.

Ibom is not intended to be project level BOM. Your issue is no different from trying to include wiring harness/terminals or fasteners or other auxiliary things into the BOM: if it doesn't have a footprint then it's not in ibom.

Intended solution for this is to export your bom to another tool that manages complex project level bom. There is a copy to buffer button at top right corner of the table and export to csv/txt functions in one of the menus.

bwack, 2023-04-08:
OK. Thanks !

closed 2023-04-08 by bwack

set-soft, 2023-04-10:
> Though, sometimes in large projects like mine, you have multiple orders per part. Say you want DIP sockets with your 74LS14 ICs, or you have this large glass fuse, but you also need the fuse holder.

In my projects I add a symbol for each thing. I.e. a LED with a LED pipe, I add an extra symbol for the LED pipe, not necessarily with a corresponding footprint. 
KiCost approach is different, read the docs, it supports multiple parts in on symbol. KiBot supports the KiCost mechanism. 
IMHO iBoM shouldn't be used as a BoM generator, but an assembly helper.

bwack, 2023-04-10:
I don't agree by the intention, but I respect the decision by the author. If it is not the intention to be the main bom I'll go with csv files instead when ordering. I'll look into KiCost, thanks.