openscopeproject/InteractiveHtmlBom #372: Sometimes Tracks are not showing up

tormyvancool, 2023-04-13:
KiCAD 7.0.1
Your Script version: 2.6.0

[image]
But there are into the project. And Gerbers files are correct.

[image]

qu1ck, 2023-04-13:
Settings in your html page are remembered in browser. If you at some point generated ibom without tracks/zones then corresponding setting will be set to disabled. If you later generate it again but include tracks/zones then you have to manually reenable tracks/zones in html to view them.

tormyvancool, 2023-04-13:
> Have you tried click on Tracks? [image]

Yes of course: then they appear. but the goal is to export in a way that **automatically** I find the checkboxes full checked. And this doesn't occur any longer.

tormyvancool, 2023-04-13:
> 

It's what I did. I enabled them and regenerated but I go them back unchecked. Then I came here to ask ...

closed 2023-04-13 by qu1ck