openscopeproject/InteractiveHtmlBom #373: Could you please add a few features that will benefit motherboard repair workers?

rabbitfix123, 2023-04-14:
I am a motherboard repair worker, this plugin is very good, it would be perfect if a few more functions are added, such as hovering the mouse over the pad to display the network name and component location information, and the ground pad can be customized to change the color Distinction, thank you for your selfless dedication...

iharshadec, 2023-04-14:
I think you are talking about feature similar to the one which comes with Apple phone repairing PC tools. This is interesting feature but I am not sure how much it can take over from the PCB data. This is really going to be really good feature.

qu1ck, 2023-04-14:
> hovering the mouse over the pad to display the network name

If you switch to nets mode and click on the pad it will highlight the whole net as well as show the net in the list.

> ... and component location information

Which component? All the components that are connected to the net? How do you suppose that information should be shown, list of coordinates and refs?

> ground pad can be customized to change the color Distinction

You want configurable colors for specific nets? 
Doable.

rabbitfix123, 2023-04-14:
> > 将鼠标悬停在键盘上以显示网络名称
> 
> 如果切换到网络模式并单击焊盘，它将突出显示整个网络并在列表中显示该网络。
> 
> > ...和组件位置信息
> 
> 哪个组件？所有连接到网络的组件？你认为应该如何显示信息，坐标列表和参考？
> 
> > 地垫可定制改变颜色区分
> 
> 您想要特定网络的可配置颜色吗？ 可行的。

Hover the mouse over any pad to display the relevant information of the pad. For example, hovering over the position of a capacitor on the positive line of the battery will display as follows:
Pin No: C1001
NetName:VBATT

It is also a very good function to customize the color matching of a specific network. For example, the color of the GND pad can be changed to gray, and the empty pad is displayed as white, so that you can quickly query it at a glance. Once again, thank you very much for providing such a great plugin, and I hope it will get better and better.

qu1ck, 2023-04-15:
Ibom doesn't store pin names but netname you can already get, not with hover but with a click.

Custom colors I can add. I'll rename this issue to be clear what it's about..

rabbitfix123, 2023-04-15:
> Ibom 不存储引脚名称，但您已经可以获得网络名称，不是通过悬停而是通过单击。
> 
> 我可以添加自定义颜色。我将重命名此问题以明确它的含义..

alright, thank you very much

wc7086, 2023-04-17:
@rabbitfix123 知道焊盘连哪里就行了，引脚名称有啥意义

commit ee812062 (2023-05-02): Configurable net colors

Fixes #373
closed 2023-05-02 by qu1ck