openscopeproject/InteractiveHtmlBom #375: Datasheet column is not shown - KiCad 7.0.1

mikrocoder, 2023-04-17:
Hi,

The datasheet column is not shown in the generated BOM and is not selectable before. I mean if you already have such **a nice HTML BOM** it would be nice if also the entered link to the datasheet is visible. It should be selectable all columns that were created in the symbols. Self created rows are usable. Only the existing line Datasheet is always missing. Is it possible to fix it?
**Thanks.**

qu1ck, 2023-04-17:
This is because kicad treats datasheet field as a "built in" instead of custom field and doesn't import it into pcb.

You can work around this by exporting netlist file from schematic and using that file as source of extra data in ibom (it will be picked up automatically if it's newer than pcb file). Then datasheet field will be selectable.

mikrocoder, 2023-04-20:
Thanks, I will give it a try.

closed 2023-04-29 by qu1ck