openscopeproject/InteractiveHtmlBom #377: Double clicking checkbox column header is not clearing the checkboxes

vardelean, 2023-04-17:
Interactive HTML BOM Version  V2.6

As per the wiki: "Double clicking checkbox column header sets/unsets all checkboxes in that column."  is supposed to clear all set checkboxes, but it doesn't always work as intended.

Most of the times, double clicking on a checkbox column header behaves like single clicks or even 2 single clicks in a row. One needs to be really fast with the double-click to clear the set boxes, and even then, the double-click does not work consistently. 

I tried changing the double-click speed option from the OS Settings - Advanced Mouse Properties, but that had no effect. 

Is there a way to get a better timing separation between single and double-clicks on a column header?

Tested with Google Chrome Version 112.0.5615.86 (Official Build) (64-bit) and Microsoft Edge Version 112.0.1722.48 (Official build) (64-bit), Windows 11 64 bits.

Interactive BOM generated with KiCad 7.0.1

Thank you!

qu1ck, 2023-04-18:
Timeout for double click is set to 200ms which is probably too fast.

To change it override fancyDblClickHandler using [custom user script](https://github.com/openscopeproject/InteractiveHtmlBom/wiki/Customization):
```
function fancyDblClickHandler(el, onsingle, ondouble) {
  return function() {
    if (el.getAttribute("data-dblclick") == null) {
      el.setAttribute("data-dblclick", 1);
      setTimeout(function() {
        if (el.getAttribute("data-dblclick") == 1) {
          onsingle();
        }
        el.removeAttribute("data-dblclick");
      }, 200);
    } else {
      el.removeAttribute("data-dblclick");
      ondouble();
    }
  }
}
```

Here change 200 to something like 300-400 for slower double click.

vardelean, 2023-04-18:
Many, many, thanks, it worked like a charm with 300ms.
All the best!

closed 2023-04-18 by qu1ck