openscopeproject/InteractiveHtmlBom #378: Export SVG?

rsoric, 2023-04-18:
Using InteractiveHtmlBom v2.3

Greetings,

great work on the project. Would it be possible to export a SVG of the board as it's shown in the HTML canvas? I have attempted, with my limited knowledge, to extract it from the canvas using some custom .js with other libraries but to no success. Could you point me in the right direction as to how to obtain the SVG data? From my understanding, it is parsed from the custom LZ-encoded JSON object and drawn on-the-fly to the canvas as the object gets moved around. So, is it possible to just get SVG somehow?

Thank you
-Rob

qu1ck, 2023-04-18:
> From my understanding, it is parsed from the custom LZ-encoded JSON object and drawn on-the-fly to the canvas as the object gets moved around.

That is correct. The original data is a mix of svg paths and custom objects, they are described in DATAFORMAT doc. All of the drawing is happening in `render.js` so in theory you could replace all canvas context2d calls with your own. Svg format and context2d api are not directly compatible though.

It will be not trivial but a significant help would be to use something like this library https://github.com/zenozeng/svgcanvas to override default canvas and make it write to svg instead. Support for various context2d apis may vary, you may have glitches with arcs and such so be prepared to roll up the sleeves and fix that lib or break open linear algebra and geometry textbook and implement transformations on your own.

rsoric, 2023-04-20:
Thank you for the information! It'll help with getting on the right track to do this. I will let you know if I end up using this method to get the SVG.

qu1ck, 2023-04-20:
I'm curious why do you need svg in particular. Ibom can render arbitrarily scaled graphics, look at how png export works. Maybe that will cover your needs.

rsoric, 2023-04-25:
We were looking to use that SVG to help us create custom parts for fritzing in a way which can be scripted. We are currently looking at some other solutions from KiCad directly, but your tool was a great help to get us started.

set-soft, 2023-04-25:
Hi @rsoric !
If you are looking to automate a workflow I'll suggest to try [KiBot](https://github.com/INTI-CMNB/KiBot).
You can generate iBoM files and you can filter/modify components and their 3D models.
KiBot can generate SVG plots and is intended for automating the generation of documentation and fabrication files.
You can also add outputs if you know Python (they act as plugins) and if you don't know about it you can propose a feature.

qu1ck, 2023-04-29:
If you do end up with a working svg export let me know. I can consider merging it if it does not add any heavy libs as dependencies. At the very least I can mention it on wiki.

closed 2023-04-29 by qu1ck