openscopeproject/InteractiveHtmlBom #379: Feature Request: Toggle SMD and Through Hole parts on BOM list.

JayShoe, 2023-04-19:
Hello, 

I appreciate all the hard work you've put into this!

I've been printing the Interactive BOM list, taping the parts to the paper, and then I assemble it from that. It works great! Assembly is a two part process, one for SMD, and then another assembly for Through Hole. 

I want to remove the through hole parts so I can print a copy of the BOM without them. Then, I want to hide the SMD parts and show all TH so i can gather my through hole parts without sifting through all the SMD stuff. I believe a toggle for SMD and a toggle for TH would make sense near the top right of the interface. 

Anyhow, I make due and love this plugin. Thanks again. 

Jay

Funkenjaeger, 2023-04-19:
I like it.  This should be useful for anyone using IBOM to place SMD parts for reflow soldering followed by a separate stage of hand-soldering through-hole components later.

qu1ck, 2023-04-19:
> I want to remove the through hole parts so I can print a copy of the BOM without them. Then, I want to hide the SMD parts and show all TH so i can gather my through hole parts without sifting through all the SMD stuff. 

Can you clarify what you mean by "remove" and "hide" here. Filter them out of the bom table? Not draw them in render? Both?

Generally I'm against modifying render in a way that makes it diverge from what real board looks like. Holes on the board don't disappear if you are in SMD stage of assembly. On the other hand, seeing that hole can help you navigate a busy board: "oh this tiny 0604 C65 goes just to the left of that big power diode's anode".

JayShoe, 2023-04-19:
Thanks for taking the time to reply!

I'm talking about filtering the list, not filtering the board.

I print the BOM without the image. I then use double sided tape to tape the components to the paper for assembly. Removing the through-hole parts just makes the list shorter.... So instead of printing 3 pages of parts, I might get away with 2 pages, with the through hole components removed. Then I can later filter out the SMD components and use that list to start installing the TH stuff. While assembling I keep the InteractiveHTMLBOM up on my computer to confirm part placement locations. 

[image]

qu1ck, 2023-04-19:
Yeah, that can be done. I just have to think of a way to add another toggle without bloating the toolbar. It's already at a limit of narrow/mobile screens.

gwideman, 2024-05-17:
I too am very interested in the SMD/THT filtering capability, but contrary to...

> I'm talking about filtering the BOM list, not filtering the board image.

... I would like to filter the board image as well.   This is because SMD and THT can be two very different stages in our assembly workflow, and having separate BOM lists and board images would be a great help to focus on what matters for that step.

How's thinking coming along on this possibility for your lovely InteractiveHtmlBom ?

For what it's worth, I wouldn't mind adding a field that contains a string "SMD" or "THT", but I'm pretty sure footprints have some SMD/THT attributes that you can query just as easily.

qu1ck, 2024-05-17:
I will not change board render, it's supposed to represent what the physical board looks like. 

There is a "Mark when checked" feature that you can use to highlight arbitrary group of components, if you have a custom field to denote smd/tht it will be very easy to do that. Or you can generate separate bom files for smd/tht groups using variant whitelist.