openscopeproject/InteractiveHtmlBom #381: DNP outlined components are marked green => placed

nopeppermint, 2023-04-20:
I have an issue with the DNP outlined feature:

I assembled before some components, therefore they were once marked as assembled.
Now I decided to not assemble them and but the DNP field in Kicad.

Now it looks like this:

[image]

Red => currently selected component => all good
Green => placed components, but they also have the DNP flag and are therefore not with a solid grey pad => bug 1
Some of the components => round testpoint in the bottom right corner are excluded from the ibom but still have the "placed" green color..

nopeppermint, 2023-04-20:
I would expect only components to be marked green when:

A = they are placed
B = they are not excluded via DNP
C = they are not excluded via "component blacklist" on creation

so => green = A & B & C

;-)

nopeppermint, 2023-04-20:
btw. this happens independent if I delete the old ibom file before creating or not, seem to be stored somewhere else in a cache..

nopeppermint, 2023-04-20:
the only way to fix it manually is to:
- toggle those two flags in the Kicad PCB 
[image]
- generate ibom
- mark all components as not placed
- toggle back the two flags in Kicad PCB
- regenerate ibom

nopeppermint, 2023-04-20:
and for the TP* the "component blacklist" has to be deleted, generate ibom, toggle the placed flag, but pack the component blacklist, regenerate ibom

[image]

BTW. I use InteractiveBOM v2.6.0 from the kicad package manager

qu1ck, 2023-04-20:
Checkbox state and other settings are stored in browser local storage and it is tied to board name and revision.

It is assumed that if you change the board significantly you would bump revision as well, that way settings will effectively reset. But there probably should be a manual reset button.

As a workaround you can reset settings manually by opening browser dev tools and removing entries from local storage 

[image]

or by invoking personal data removal tool in your browser (ctrl-shift-del in chrome), but this will delete a lot of stuff.

dresco, 2023-04-27:
> Checkbox state and other settings are stored in browser local storage and it is tied to board name and revision.

May I check, where is the revision taken from - the schematic page settings? Thanks...

qu1ck, 2023-04-27:
No, board page settings. Ibom uses board data for everything, only exception is extra fields that you can optionally source from netlist file or eeschema generated xml.

dresco, 2023-04-28:
> No, board page settings. Ibom uses board data for everything, only exception is extra fields that you can optionally source from netlist file or eeschema generated xml.

Ah yes, got it. Thank you!

qu1ck, 2023-05-03:
Ok, there is now a reset button allowing you to start from a clean slate.

closed 2023-05-03 by qu1ck

nopeppermint, 2023-05-15:
thanks @qu1ck  would be nice if you could also create a new Kicad Plugin release version