openscopeproject/InteractiveHtmlBom #383: Component names not generating from Eagle board

kvandekrol, 2023-04-28:
I generated an IBOM from a medium-sized board, around 200 components. Everything works great in Chrome except that only about 20% of the components have labels. This particular .brd file has 4 sub-boards, and one of the smaller boards has all of its components named, but the other boards are spotty with only a handful of components labeled across all three. All other layers are OK, it's just the component names that are affected.

IBOM file attached for reference. It was generated with Python 3.10 on Windows 10. Repo installed this morning, so latest versions of everything. No errors during installation or IBOM generation.

I should also mention that it was created from a .brd designed in Eagle 7.7.0. I have a perpetual license and don't plan to upgrade to the Autodesk version anytime soon. If that's the cause, or makes troubleshooting difficult, then please disregard!

[ibom.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/11355360/ibom.zip)

qu1ck, 2023-04-28:
It would help if you could provide a board file. Strip out any sensitive parts if you want, leave just enough to be able to show the problem.

kvandekrol, 2023-04-28:
I tested it with 3 or 4 other .brd files and had the same result for each, so here's a much simpler one that shows the same issue -

BRD: [bb.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/11355910/bb.zip)
IBOM: [ibom.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/11355914/ibom.zip)

qu1ck, 2023-04-28:
I fail to see the issue here. First of all what is the "component name" or "label" that you are expecting? The bom from first post looks perfectly fine to me with all fields populated.

Bom in your second post has some diodes that don't have values. Looking at the source board file, it's missing there too so ibom worked with the data it had:

Diode with a value:
```
<element name="D1" library="aion-diodes" package="DO41-3" value="1N5817" x="13.335" y="-13.335"/>

```
and one without:
```
<element name="D2" library="aion-diodes" package="DO35-2" value="" x="12.7" y="16.51"/>
```

qu1ck, 2023-04-28:
Ah, you probably mean silkscreen markings. Most text will not show up because eagle font data is not supported but references should show up. I'll look into that.

qu1ck, 2023-04-28:
Yeah that must be it. @Funkenjaeger I think parser can be improved to look into library definition for ">NAME" text if there is no explicit "NAME" sub attribute. Do you want to tackle that?

kvandekrol, 2023-04-28:
That did it. I smashed all of the components and now everything's showing up as expected -

[ibom.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/11356198/ibom.zip)

Funkenjaeger, 2023-04-28:
You beat me to it, I just arrived at the same conclusion regarding smashed components as you posted it!

My memory is fuzzy on the details, but I think that "smashing" is a concept that Autodesk buried when they took over Eagle.  When I opened your board file in Eagle 9.6.2 and re-saved it, in the new file all the components had been automatically smashed.  

I could probably handle this by reverse-engineering whatever logic they use to decide where to position the NAME and VALUE attributes when you smash a component.  At a glance, it looks like NAME is trivial (same position & orientation as the component itself), but I'd need to poke at it a bit more to figure out how it's choosing the location for VALUE if I wanted to match it precisely.  

Is that an important enhancement in your mind, or is smashing it yourself an acceptable workflow for you?

qu1ck, 2023-04-28:
This is diode package from board file:
```
<package name="DO41-3">
<description>&lt;b&gt;DO-41 0.3" pad spacing&lt;/b&gt;</description>
<wire x1="2.032" y1="-1.27" x2="-2.032" y2="-1.27" width="0.1524" layer="21"/>
<wire x1="2.032" y1="-1.27" x2="2.032" y2="1.27" width="0.1524" layer="21"/>
<wire x1="-2.032" y1="1.27" x2="2.032" y2="1.27" width="0.1524" layer="21"/>
<wire x1="-2.032" y1="1.27" x2="-2.032" y2="-1.27" width="0.1524" layer="21"/>
<wire x1="3.81" y1="0" x2="2.413" y2="0" width="0.762" layer="51"/>
<wire x1="-3.81" y1="0" x2="-2.54" y2="0" width="0.762" layer="51"/>
<pad name="C" x="-3.81" y="0" drill="1.1" diameter="1.778" shape="square"/>
<pad name="A" x="3.81" y="0" drill="1.1" diameter="1.9304"/>
<text x="0" y="0" size="1.016" layer="25" font="vector" ratio="14" align="center">&gt;NAME</text>
<text x="-1.905" y="-2.794" size="1.27" layer="27" font="vector" ratio="12">&gt;VALUE</text>
<rectangle x1="-1.651" y1="-1.27" x2="-1.143" y2="1.27" layer="21"/>
<rectangle x1="-2.413" y1="-0.381" x2="-2.032" y2="0.381" layer="21"/>
<rectangle x1="2.032" y1="-0.381" x2="2.413" y2="0.381" layer="21"/>
</package>
```

Note the text elements, they have their own coordinates. Probably angle too if it's not 0? @kvandekrol can you generate a board file with a footprint that has rotated text? Put it on board not smashed.

kvandekrol, 2023-04-28:
Yep, here you go:

[bb.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/11356833/bb.zip)

RX5 on the right-hand side of the board is vertical. D2, D3, D4 and D5 all have the "Spin" attribute checked, which means the labels for D4 and D5 are upside down while D2 and D3 are right side up. All are unsmashed (though a handful of other components are smashed). Let me know if that's enough to go off of.

kvandekrol, 2023-04-28:
> Is that an important enhancement in your mind @kvandekrol, or is smashing it yourself an acceptable workflow for you?

It would definitely be a time-saver. It would be worth the extra effort on my end if it's the only way since the outcome itself is so useful, and it's certainly not a daily or even weekly workflow thing... but if it's an easy enough fix, I know there are a lot of people still on pre-Autodesk Eagle for the same reason.

> At a glance, it looks like NAME is trivial (same position & orientation as the component itself), but I'd need to poke at it a bit more to figure out how it's choosing the location for VALUE if I wanted to match it precisely.

I believe this is defined in the component footprint. I can place NAME and VALUE wherever I want when designing the component, and once smashed they'll keep the exact coordinates that the part had as default. For my libraries, NAME is centered to the component maybe 80% of the time, but in a lot of cases like potentiometers, transistors and toggle switches it would overlap a pad if it was centered. 

I typically don't use VALUE but would guess it could be ignored if it's not currently being handled. It would definitely be better than showing it by default. I know most of my components have a value label defined, but I have never paid attention to it and I think it's usually positioned the same as the NAME.

Funkenjaeger, 2023-04-28:
> 

> I believe this is defined in the component footprint. I can place NAME and VALUE wherever I want when designing the component, and once smashed they'll keep the exact coordinates that the part had as default.

You're quite right, now I just need to make sure I handle rotation (of the text within the footprint, plus rotation of the component itself) correctly.  I just installed Eagle 7.7.0 (what a throwback!) to let me play around with test cases, so I'll poke at it and hopefully push a fix soon.

Funkenjaeger, 2023-04-29:
Created PR #384 for this.  @kvandekrol please give the fix a try on your various other board designs.
For the example you shared, it looks OK to my eye:
[image]

kvandekrol, 2023-04-29:
Almost there! The one thing I noticed is it doesn't take into account the "Mirrored" attribute of unsmashed components that have been moved to the bottom side of the board and the NAME label still prints on the top side. Here's an edited version of the .brd file that has four resistors (RX2, 3, 4, 6 on the right-hand side of the board) set to Mirrored. The top two (RX3, RX6) are unsmashed and the bottom two (RX2, RX3) are smashed.

[bb_mirrored.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/11359569/bb_mirrored.zip)

And here's the IBOM:
[ibom.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/11359570/ibom.zip)

Funkenjaeger, 2023-04-29:
Alright - I pushed an update that handles mirroring now, give it another go.
[image]

kvandekrol, 2023-04-29:
Perfect. I tested it on a different board with several mirrored components of all types and it worked flawlessly!

qu1ck, 2023-04-29:
Thanks @Funkenjaeger for continued support.

closed 2023-04-29 by qu1ck