openscopeproject/InteractiveHtmlBom #386: EasyEDA已经不支持导出json文件了，所以我应该怎么使用这个项目还是直接抛弃

mingtianshi, 2023-05-05:


qu1ck, 2023-05-05:
Hope this translation is correct

[image]

I just checked and you can still export json so I am not sure what you mean.

[image]

mingtianshi, 2023-05-05:
[image]
I am using the web version, and in fact, the same applies to using the ezyeda professional version.

qu1ck, 2023-05-06:
That looks like EasyEDA Pro which is not supported because file format is not documented.

But it has built in soldering assistance tool with similar functionality

[image]

closed 2023-05-08 by qu1ck