openscopeproject/InteractiveHtmlBom #388: AttributeError: 'BOARD' object has no attribute 'GetModules'

mondalaci, 2023-05-21:
I've recently upgraded to KiCad 7.0.2, and upon trying to generate the BOM, the following dialog appears.

[image]

qu1ck, 2023-05-21:
You are running a very old version, update your plugin.
If you are running from kicad and not from terminal then just delete your old version and install from kicad's plugin manager.

mondalaci, 2023-05-22:
I'm running the latest version, but I can't reproduce the issue anymore, so I'm unsure what was happening.

Thanks for the quick reply, and keep up the great work!

closed 2023-05-22 by mondalaci

qu1ck, 2023-05-22:
Judging from the paths in the error above you have ibom in your home dir, that is not standard location where plugin manager installs things.

If you run into this issue again then make sure to delete your manual installation folder and just install from plugin manager.