openscopeproject/InteractiveHtmlBom #389: Feature Request: Reset Button for Placed/Sourced Components

jakeradler, 2023-05-26:
I frequently use the same IBOM when filling out multiple boards simultaneously, and manually resetting the checkboxes feels unnecessary. Ideally, The feature would be a button with a JS script that turns all the checkboxes (2nd and 3rd cell elements in bomrow) false or directly deals with the logic in ibom.js.

jakeradler, 2023-05-26:
I was able to disable the checkboxes through a js script but there appears to be some backend logic (don't have the time to immediately explore.

Ex:
`function uncheckBomRowCheckboxes() {
    var bomRow = document.getElementById('bomrow1');
    if (bomRow) {
        var cells = bomRow.getElementsByTagName('td');
        if (cells && cells.length > 2) {
            // uncheck the checkbox in the second cell
            var checkbox1 = cells[1].querySelector('input[type="checkbox"]');
            if (checkbox1) {
                checkbox1.checked = false;
            }
            // uncheck the checkbox in the third cell
            var checkbox2 = cells[2].querySelector('input[type="checkbox"]');
            if (checkbox2) {
                checkbox2.checked = false;
            }
        }
    }
}

// Call the function
uncheckBomRowCheckboxes();
`

qu1ck, 2023-05-26:
Double clicking the checkbox header will toggle all checkboxes, you can reset them that way.

There is also a reset config button added recently in 42a02270ff7d0ecc7704fc110a3af57247ce0cc1 (not in released version yet).

closed 2023-05-26 by qu1ck