openscopeproject/InteractiveHtmlBom #390: Feature Request: Placed parts are rendered differently

hacsccc, 2023-06-01:
While placing parts, it would be nice to have a visual feedback on the board-render which parts are already placed.
I propose to add a CSS class 'placed' that gets toggled once the placed checkbox is checked. Parts that are placed get rendered in a lighter tone.

Then, the user can see where on the PCB parts are not yet placed by looking at the canvas.

cpresser, 2023-06-01:
That was my throwaway account. :facepalm:  I am commenting here to get proper notifications.

qu1ck, 2023-06-01:
Also already done, in the settings menu you can switch which checkbox will be affect the render highlight.

[image]

Rows in the table don't need a highlight, checkbox itself is enough.

closed 2023-06-01 by qu1ck