openscopeproject/InteractiveHtmlBom #393: Getting Syntax Errors

ankithjv, 2023-06-07:
File "Z:\Git\InteractiveHtmlBom\InteractiveHtmlBom\generate_interactive_bom.py", line 84, in <module>
    main()
  File "Z:\Git\InteractiveHtmlBom\InteractiveHtmlBom\generate_interactive_bom.py", line 63, in main
    os.path.abspath(args.file), config, logger)
  File "Z:\Git\InteractiveHtmlBom\InteractiveHtmlBom\ecad\__init__.py", line 15, in get_parser_by_extension
    return get_generic_json_parser(file_name, config, logger)
  File "Z:\Git\InteractiveHtmlBom\InteractiveHtmlBom\ecad\__init__.py", line 35, in get_generic_json_parser
    from .genericjson import GenericJsonParser
  File "Z:\Git\InteractiveHtmlBom\InteractiveHtmlBom\ecad\genericjson.py", line 107
    f' "{s}" that are present in text drawing'

qu1ck, 2023-06-07:
What file format are you trying to generate bom for?

Can you copy full error message, it is cut off at the bottom.

ankithjv, 2023-06-07:
it is a json file that was converted from an Allegro brd file using the script you provided 

Traceback (most recent call last):
  File "Z:\Git\InteractiveHtmlBom\InteractiveHtmlBom\generate_interactive_bom.py", line 84, in <module>
    main()
  File "Z:\Git\InteractiveHtmlBom\InteractiveHtmlBom\generate_interactive_bom.py", line 63, in main
    os.path.abspath(args.file), config, logger)
  File "Z:\Git\InteractiveHtmlBom\InteractiveHtmlBom\ecad\__init__.py", line 15, in get_parser_by_extension
    return get_generic_json_parser(file_name, config, logger)
  File "Z:\Git\InteractiveHtmlBom\InteractiveHtmlBom\ecad\__init__.py", line 35, in get_generic_json_parser
    from .genericjson import GenericJsonParser
  File "Z:\Git\InteractiveHtmlBom\InteractiveHtmlBom\ecad\genericjson.py", line 107
    f' "{s}" that are present in text drawing'
    ^
SyntaxError: invalid syntax

ankithjv, 2023-06-07:
switched to python 3.6 and am getting this error now

ankithjv, 2023-06-07:
Is the width just not allowed to be 0?

qu1ck, 2023-06-08:
> which version of python should I be using?

Python 3.6+, 3.10 or later is recommended.

> as wxpython is not available on 3,10 so i went back to 3.5

Not sure what you mean, it is available even on 3.11

> 2023-06-07 11:14:29,962 ERROR File Z:\A1.json does not comply with json schema. {'type': 'segment', 'start': [4440.0, -3935.0], 'end': [4010.0, -3935.0], 'width': 0.0} is valid under each of {'$ref': '#/definitions/DrawingRect'}, {'$ref': '#/definitions/DrawingSegment'}

This doesn't make sense. Can you provide the json file? You can remove any pcb objects you want if you can't share proprietary design, just leave bare minimum that still shows the issue.

qu1ck, 2023-06-08:
> Is the width just not allowed to be 0?

It is allowed to be 0.

ankithjv, 2023-06-08:
nevermind sorry I got it to work. Thank you! Is there anyway for me to change code to get more than just the front and back of the board? Get the other layers too?

ankithjv, 2023-06-08:
I was running it through anaconda shell. And I was downloading all libraries and packages through anaconda (hence I couldnt get a wxpython on newer versions of python). Once I just pip installed and ran on command prompt everything was fine. Python 3.11.

ankithjv, 2023-06-08:
So I couldn't add anything to your python code to support inner layers?

qu1ck, 2023-06-08:
> So I couldn't add anything to your python code to support inner layers?

The tool is open source, of course you can if you have the skills and will to implement it. But you have to adjust the skill script either way and I suggested a simple workaround of merging the layers first, that way you will not have to touch any of ibom's logic.

If you want to contribute your changes and submit a PR make sure that it is generic enough and will work for other supported CAD packages. I can add kicad implementation later myself.

If you don't plan to contribute, do whatever you want, code is MIT licensed. Ask more specific questions if you need pointers with the code. 

For starters read the DATAFORMAT.md to understand how objects are represented in ibom. The skill script generates json in a format that complies with schema very close to the format described there. If you want to add new layers, then in broad strokes you would need to:
1. Extend the json schema to support other layers
2. Extend the data format to support other layers
3. Implement changes to skill script (outside of this repo)
4. Make sure new layer data gets into generated page javascript in pcbdata global variable
5. Implement the changes to javascript renderer to draw stuff on the background canvas (or add new canvas)
6. Add toggles and settings to show/hide your layers in settings menu

If you look at how things work for, say, silkscreen layer and copy that, it would get you mostly there.

qu1ck, 2023-06-09:
Yes. At that point you would probably want configurable colors for layers too.

closed 2023-06-12 by qu1ck

ankithjv, 2023-06-12:
Sorry for adding extra issues, this is a very impactful project. Am going to try to add support for inner layers.

ankithjv, 2023-06-14:
@juulsA updated the Allegro to json conversion to incorporate inner layers so I will work on adding support to this now. Will let you know how it goes. Thanks again