openscopeproject/InteractiveHtmlBom #397: Unable to process recent (7.99) Kicad files:  module '_pcbnew' has no attribute 'PCB_FP_TEXT_T'

sborshch, 2023-06-28:
Kicad files:  module '_pcbnew' has no attribute 'PCB_FP_TEXT_T'
Kicad version: 7.99.0-1663-g06c8185133, release build

sborshch, 2023-06-30:
in another board file:

File "/usr/lib/python3/dist-packages/pcbnew.py", line 12004, in <module>
    PAD_SHAPE_RECT = _pcbnew.PAD_SHAPE_RECT
AttributeError: module '_pcbnew' has no attribute 'PAD_SHAPE_RECT'

Kicad version: 7.99.0-1689-g2ae646136e, release build

qu1ck, 2023-06-30:
Did you self compile kicad or installed it from a repo?

Are you running the plugin from command line or from pcbnew as a plugin?

sborshch, 2023-06-30:
Self-compiled. 

Runing from command line with 
```
python3 ~/opt/kicad/share/kicad/plugins/InteractiveHtmlBom/InteractiveHtmlBom/generate_interactive_bom.py \
 --hide-silkscreen                      \
 --checkboxes ""                        \
 --bom-view top-bottom                  \
 --layer-view F                         \
 --dest-dir .                           \
 --name-format %f_bom                   \
 --sort-order DD,DA,U,VT,VD,A,R,C,L,FB,~,ZQ,FU,JMP,JP,XP,XS \
 --show-fabrication                     \
 --include-tracks                       \
 --netlist-file ${!#%.kicad_pcb}.net    \
"$@"
```

qu1ck, 2023-06-30:
Can you post full stack trace?

Did you set ``$LD_LIBRARY_PATH` and `$PYTHON_PATH` to include pcbnew.py and pcbnew.so that correspond to nightly build? By default it will not get into standard paths so you either have to correct cmake install paths or set those env vars manually for nightly.
Those errors kinda look like you are using pcbnew.py with not matching pcbnew.so.

sborshch, 2023-06-30:
Sorry, for some reason there is pcbnew.py from april, 17 in my /usr/lib. I'll try to rebuild Kicad completely

sborshch, 2023-06-30:
Sorry. problem was with my kicad build and installation script.

closed 2023-06-30 by sborshch

qu1ck, 2023-06-30:
No worries, glad you got it working.