openscopeproject/InteractiveHtmlBom #398: Unrelated components erroneously grouped together, and 'cached' values

jwilo, 2023-06-29:
I've witnessed this several times now, where components with different values, and a different part number get grouped together. Often this occurs in parallel with the value field shown in the ibom, different to what is shown in KiCad.

For example, I have R15 & R16, with a value of 62R, and an MPN of ERJ-U1TF62R0U. In addition, R79 and R80, value of 430R and MPN of ERJ-U03F4300V.

ibom groups them together as below. Note that the 430R resistor is shown as 420R - which it did **used** to be in this project, but isn't any more - it is as if a cached dataset is somehow being used.

<html>
<body>

R15, R16, R79, R80 | 62R, 420R |   | 4
-- | -- | -- | --

</body>
</html>

qu1ck, 2023-06-29:
You should see values grouped together like that with comma only if you did not tick "group" for values in fields tab in config.

Also remember to update pcb from schematic and save it if you changed field values.

jwilo, 2023-06-29:
I must be misunderstanding usage of 'Group'.

I have Value set to Show, and MPN set to Show **and** Group. Logically, I would not want to group anything on value together; for example, what if we had a 0402 size 10K resistor, and an 0805 10K resistor - they would have the same value, but would be of course different due to their MPN and thus we'd only want to group by MPN.

Regardless, I tried your suggested and have ticked Show and Group for both Value and MPN.

Now, the 62R and 420R resistors show on separate lines - but with empty MPN fields. Note also, 420R is still shown in the value, not the correct 430R.

PCB is fully up to save, all saved etc, KiCad opened/closed/PC restarted.

jwilo, 2023-06-29:
I've also tried grouping on Value, and not MPN too - but this gives me plenty of issues due to the above. For example, 10K resistors and 10K NTCs are now grouped together.

jwilo, 2023-06-29:
Hmm, seems I've just cleared the issue - I deleted the MPN fields, saved. Updated PCB, generated ibom and as expected MPN values were empty.

Put the offending MPNs back in place, saved, refreshed PCB and generated ibom and all was OK.

I've had a couple of issues with MPNs before, where they've been copy/pasted from Mouser website/invoices etc, and non-printable characters, carriage returns etc have been saved in the KiCad fields. Clearly without knowing how your tools processes these fields however, I can't see how/why this could possibly cause an issue.

qu1ck, 2023-06-29:
The way grouping works is that combination of all fields marked with "group" form a key. Parts that have matching grouping key are combined. So if you have same MPN but different values and group only by MPN then those different values will be grouped together.

Another thing to pay attention to: which file is used for extra fields data, you see it at the top of fields tab in ibom. It can be a netlist file, eeschema bom xml file or kicad_pcb file. Ibom will pick the one with latest modification date (unless you manually point it to another one) so usually it works out but if for example you have some other netlist file in the project folder with different values it may mess up your bom. It can also happen if you have multiple boards in same folder.

Regarding non printable characters, if you can put together a minimal reproducible example I can investigate further.

jwilo, 2023-07-04:
Ah, this may be it then - how do I manually point to a specific file? Is this the 'Extra data file' field under the 'Fields' tab? It is currently set to the kicad_pcb file. When I edit values/MPNs etc in a KiCad schematic and save it, which file does this edit?

Ref the non printable chars, ignore me on this one - apologies - that was some irrelevant ramblings, the issues we've seen have been with some of other connected systems; ERP etc, not the generation of the ibom.

qu1ck, 2023-07-04:
> Is this the 'Extra data file' field under the 'Fields' tab?

Yes

> how do I manually point to a specific file? 

Click the `...` button next to it and select another file but you don't normally have to do that, ibom will always automatically pick correct one if you keep your kicad_pcb updated.

> When I edit values/MPNs etc in a KiCad schematic and save it, which file does this edit?

Schematic. But when you do "update pcb from schematic" it pulls that data from schematic into pcb so always do that before running ibom.

qu1ck, 2023-08-30:
Cleaning up unactionable issues.