openscopeproject/InteractiveHtmlBom #399: Sheet path/name no longer available for export

Qix-, 2023-06-30:
HI there, firstly great plugin that has helped me save a bunch of time, so thank you so much for making it!

3-4 months ago I remember being able to export the sheet path/name as fields for grouping, but that no longer seems to be the case. When testing portions of the board, it was really helpful as the schematic had individual components broken out into sheets. Now it seems that functionality was removed, or I've done something to somehow disable it.

Is there a way to re-enable sheet names/paths, or was that removed from the plugin?

qu1ck, 2023-06-30:
It wasn't removed but if you are using nightly the API changed and needs a compatibility fix.

Which kicad version are you using and which file are you using for extra fields? pcb or netlist or xml?

Qix-, 2023-06-30:
This is 7.0.0 release build. I'm just clicking the button from the toolbar in the PCB editor and not changing any files.

qu1ck, 2023-06-30:
Check which file shows up in the extra data file field

[image]

By default ibom will pick supported file with latest modification date, supported files are .kicad_pcb, .net and .xml. But not all of them have support for sheet fields, if I remember correctly only .kicad_pcb provides those fields out of the box. So if you have a netlist file with later modification date it will be picked and sheetname fields will not show up. Make sure you update kicad_pcb and save it before running ibom.

Qix-, 2023-06-30:
You nailed it, it was pulling an XML instead of the .kicad_pcb file. Saved the PCB and now the sheetfiles are showing up.

Thanks a lot!

closed 2023-06-30 by Qix-