openscopeproject/InteractiveHtmlBom #400: Fields display corruption

rgalland, 2023-07-14:
[image]
This has happened since switching to Kicad 7.0. I have components that have a certain number of custom fields and it seems that iBom cannot cope with it. I ran some tests with a simple BOM and worked out that if I remove enough fields, the corruption disappears. Basically, the number of custom fields should not exceed 9. This is not something that used to happened when I used Kicad 6 and probably an older version of iBom.

qu1ck, 2023-07-15:
I believe this was fixed in 3302982ea7918ce4544ec6c3b9f0ce0350b74f23

Can you try it? You will have to uninstall the plugin from kicad's plugin manager and follow manual installation instructions in wiki.

rgalland, 2023-07-15:
Excellent, thank you for your quick response. I fixed the lines manually in the code for now and it works fine.

closed 2023-07-15 by rgalland