openscopeproject/InteractiveHtmlBom #401: Thread two HTML Boms together

ankithjv, 2023-07-17:
Could I create a bom in which I have two boms next to each other. Side by side in the view?

qu1ck, 2023-07-17:
Not sure what you are asking. Just 2 bom tables from different boards, no renders? Generate 2 separate boms and then copy paste the table into excel or google sheets.
If you are asking for something else, add more details.

ankithjv, 2023-07-17:
I mean can I merge two html files together so one ibom will have two renders in the screen just next to each other in one html file.

qu1ck, 2023-07-17:
So you want one common bom table for 2 different boards? No, there is no functionality like that but you can merge your boards in your cad and then generate a bom.

ankithjv, 2023-07-18:
okay! Thanks!

closed 2023-07-18 by ankithjv