openscopeproject/InteractiveHtmlBom #402: Have Netlist and BOM stacked on top of eachother

ankithjv, 2023-07-19:
Is there anyway to have both the netlist table and bom tables show at the same time just stacked on top of each other? Or where I would add that functionality?

qu1ck, 2023-07-19:
Such view is not supported. If you want to implement it you will have to essentially create 2 tables. Look at `populateBomHeader()` and `populateBomBody()` and how they change the output based on `settings.bommode`.

ankithjv, 2023-07-19:
okay thanks where in the code are the tables actually being rendered and called?

qu1ck, 2023-07-19:
The functions I mentioned do the actual rendering of the table. They render it into predefined table element that exists in the html template. Look for `<table class="bom" id="bomtable">` in `ibom.html`. One way to achieve your goal is to modify the template to have 2 such tables and pass the id of the element to populate functions along with bom mode.

ankithjv, 2023-07-20:
what would the ids of the elements be?

ankithjv, 2023-07-20:
would I keep "placeholderColumns" as null?

qu1ck, 2023-07-20:
Yes, that is part of column dragging functionality, they are null when not dragging.

ankithjv, 2023-07-20:
Waht is placeholderelements?

ankithjv, 2023-07-20:
how would I pass the id of the element?

qu1ck, 2023-07-20:
> Waht is placeholderelements?

Empty elements that get rendered as a placeholder for column that is being dragged.

> how would I pass the id of the element?

As a function parameter? If you are asking such basic questions you may want to go through a javascript tutorial or two first. 

Right now ibom just stores a global var referencing the elements for bom header and body and populate functions render the data into those elements. Since you will have 2 tables instead of one you should pass the target element or it's id into the function so it knows where to render.

ankithjv, 2023-07-20:
basically I was wondering where the table id was being referenced inside the function. I see that in table util that you establish it bom as the table id. also once I do this where do I get rid of the menu toggle to switch between modes of the table?

qu1ck, 2023-07-20:
You can just remove the buttons from the html template.

ankithjv, 2023-08-28:
So I implemented the tables int he way that you suggested where both get rendered in the functios but now I am having the issue where, if there are only 100 components and 200 nets then the first 100 values will always only reference the component table. Is there anyway to make the tables behave independently but connect in terms of a component tied to a net will show the component and the nets on the other table. Or maybe one table with all of the information.

qu1ck, 2023-08-28:
I don't understand your issue, it sounds like a bug in your implementation.

closed 2023-10-23 by qu1ck