openscopeproject/InteractiveHtmlBom #403: EasyEDA, no traces are shown

vanBassum, 2023-07-21:
Hello,

I just made a BOM from a project of mine. But it so happens that there are no traces visible. I did so some searching, but couldn't find anything.

The command used:
`python generate_interactive_bom.py C:\Users\bas\Desktop\bldc\pcb.json`

The output:
```
Loading C:/Users/Bas/Desktop/bldc/pcb.json
2023-07-21 13:41:17,515 INFO EasyEDA editor version 6.5.29
2023-07-21 13:41:17,530 INFO Compressing pcb data
2023-07-21 13:41:17,585 INFO Dumping pcb data
2023-07-21 13:41:17,589 INFO Created file C:\Users\Bas\Desktop\bldc\bom/ibom.html
2023-07-21 13:41:17,589 INFO Opening file in browser
```

Here is the JSON and the result
[bldc.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/12123246/bldc.zip)

qu1ck, 2023-07-21:
To make traces visible include the `--include-tracks` on the command line. To see all config flags run the script with `--help` or read [wiki](https://github.com/openscopeproject/InteractiveHtmlBom/wiki/Usage#command-line-options)

vanBassum, 2023-07-21:
Ah that was an oversight on my part. I also found this: https://github.com/turbobabr/easyeda-ibom-extension
Making live simpler. Thanks for the support.

closed 2023-07-21 by vanBassum