openscopeproject/InteractiveHtmlBom #407: Feature request: Projector mode (fully black + very specific scaling)

Xasin, 2023-08-18:
Hi! 

This plugin has already been a great help to me a few times, and I really enjoy how easy it makes it to find component locations.

I do think that there might be one fairly easy way to make this tool even more useful: By giving it a "projector mode" 

If the background could be set to a fully black colour, and the zoom of the front/back view could be fixed to a specific, predefined factor (via a numeric input field, and by disabling zooming with the mouse), it would be possible to project the HTML Bom onto a PCB workbench surface with a calibrated (through trial and error probably) scaling factor, in a way that the BOM representation of a PCB lines up with a physical PCB that is being worked on.

Skew and other visual misalignment can usually be corrected through settings in the projector itself, so that is not something this HTML BOM would have to worry about.

This should not be too much effort, but could be extremely helpful for PCB assembly processes and debugging.
Once the scaling factor is calibrated, setup would be as simple as loading up the HTML BOM in a browser and throwing it onto the projector, then taking care to manually line up the physical PCB with the projected image as needed.

A small but more complex extra would be hiding or minimizing the BOM list during "projector mode", instead only showing the information of the currently marked component(s), and stepping through them using a keyboard or remote.

I might look into the code and see how complex adding it would be myself, once the current heatwave is over; I am opening this issue to allow for discussion on the details before digging in, and perhaps seeing if someone else has attempted something similar.

There are, for example, these videos of working proof of concepts, though neither of them seem to have added their contributions to this repository:

https://www.reddit.com/r/KiCad/comments/emzwkm/demo_projection_mapping_the_interactivehtmlbom/

https://www.youtube.com/watch?v=MRn1CRluOb0

qu1ck, 2023-08-18:
You can tweak colors as you wish using [customization](https://github.com/openscopeproject/InteractiveHtmlBom/wiki/Customization) and overriding `:root` and `.dark.topmostdiv` rules in `user.css`

Regarding precise scaling it is doable but I think in practice you have to have some kind of adjustable mount for the projector, some kind of swiveling arm that you can bring in closer/further, move into and out of the workspace. It would be much easier to rely on that mount for adjusting the "fit" of the projected image than trying to guess the precise scaling number in ibom.

But this sounds like a good use case for template support (#234), having a separate ibom template optimized for projection would be nice.

Xasin, 2023-08-19:
@qu1ck the adjustability of the projector won't be of any use if the scaling of the PCB itself on the screen can't be set to a predefined value.

The way the HTML Bom seems to work right now is that it will auto-fit the PCB view, and it's all too easy for a user to accidentally scroll/zoom in the view as well. Both of those would missmatch the scaling and require a bit of readjusting.

At the very least a scaling lock would be needed so that, once adjusted, the scaling is consistent.
But I agree that most of the finer adjustments should be done with the projector mount.

qu1ck, 2023-08-19:
> the adjustability of the projector won't be of any use if the scaling of the PCB itself on the screen can't be set to a predefined value.

I don't see why. I meant physical placement adjustment like something regulating the distance between the projector and the desk.

> it's all too easy for a user to accidentally scroll/zoom in the view

Right click in the pcb view resets the zoom and pan. By adjusting the split size, window tab you can bring the default zoom to close to desired level and do the rest with projector adjustment.

I understand it's not ideal, better support in software would make it easier. Just throwing out ideas that can be used as workarounds.

Xasin, 2023-08-21:
Right click on the PCB view resets the zoom and pan *in a way that maximizes the PCB*
This means that the scale of the PCB after right-clicking is highly dependent on the window sizing and PCB size, and in no way repeatable.

I did not open the issue to ask for potential workarounds, as I am aware of the ability to manually fiddle the scale into a correct value.
I opened this issue specifically to provide a means to *avoid* this exact kind of fiddling, and to provide a repeatable scaling method to make this part of a repeatable, easy to use soldering workflow. A reliable scaling mechanism would make it much easier and more worthwhile to use the HTMLBom + Projector on a PCB workstation.

KataFF, 2023-11-05:
Just to chip in - have you heard of the Augumented Reality Debug Workbench? It uses InteractiveHtmlBom with a projector, maybe that's what you're looking for. See:
[3 minute video demo on YouTube](https://youtu.be/RbENbf5WIfc)
[GitHub source repository](https://github.com/ubicomplab/ar-debug-workbench)