openscopeproject/InteractiveHtmlBom #408: Wrong parts selected and grouped

jrzwee, 2023-08-18:
When exporting the iBom with the command line on the latest EAGLE version, I get this output:

[image]

Random parts are grouped and if I click on one part in the viewer it is the wrong part selected.

E.g.: I click on an IC and the table tells me it is a resistor.

qu1ck, 2023-08-18:
Can you provide an example board that shows the issue?

Funkenjaeger, 2023-08-20:
I'm happy to take a look as soon as @jrzwee can provide an example .brd file that reproduces the issue.

qu1ck, 2023-08-21:
Thanks, I see the issue with this board. Haven't delved deep yet, but this is not a recently introduced bug at least, probably eagle parser doesn't handle some case.

Funkenjaeger, 2023-08-21:
Thanks for sharing the example, I'll take a look.  I wouldn't be surprised if it's related to assembly variants, that's a feature I haven't used much so there may well be some surprises that the parser doesn't yet handle.

Funkenjaeger, 2023-08-22:
Okay, so the problem was related to assembly variants, but only tangentially.  Assembly variants are the only way to natively accomplish DNP in Eagle/Fusion 360, and the way I was handling those DNP's was wrong, resulting in the indexing of components to footprints being all messed up, thus the incorrect highlighting.  PR incoming.

Funkenjaeger, 2023-08-22:
Submitted #409 

@jrzwee can you please give it a try from my fork and confirm whether it's fixed this issue on your real board?

jrzwee, 2023-08-22:
You guys are great! I can confirm it is working now! @Funkenjaeger

jrzwee, 2023-08-22:
What is still a problem: the copper is missing when exported.

qu1ck, 2023-08-22:
It is off by default. Add `--include-tracks` and you probably also want `--include-nets` flags to your command.

qu1ck, 2023-08-22:
@Funkenjaeger on the test board above generating zones appears a bit broken, they don't have any cutouts for traces/vias etc. I can't recall if they were ever fully supported, is there a bug or something preventing correct holes processing?

jrzwee, 2023-08-22:
> It is off by default. Add `--include-tracks` and you probably also want `--include-nets` flags to your command.

Unfortunately adding the flags seems to lead to the same result.

Funkenjaeger, 2023-08-22:
> @Funkenjaeger on the test board above generating zones appears a bit broken, they don't have any cutouts for traces/vias etc. I can't recall if they were ever fully supported, is there a bug or something preventing correct holes processing?

Unfortunately that's due to the fact that Eagle handles zone cutouts internally while rendering, so the geometry of the cutouts is not embedded in the BRD file, and therefore it goes beyond merely parsing the file.  What's in the file is just the geometry of the zone (sans cutouts) and the set of design rule parameters which it uses to determine the cutouts.  
In fact, evidencing this, when you first open up a BRD in Eagle, the zones are rendered as unfilled polygons with a dashed outline:
[image]
And only after hitting 'Ratsnest' does it actually calculate and render the copper:
[image]

So, I think to address this I'd either have to do it in the parser (implement similar logic as Eagle does in its rendering, transform each zone into one or more much more complex polygons), or do it during rendering (in javascript).  The former seems like it would be a huge headache, the latter seems maybe feasible (composite operations in canvas?) but it would mean Eagle-specific changes residing outside the Eagle parser.  Of course, I welcome your thoughts on whether there's an easier way, etc.  We could put it in as a wishlist/enhancement issue if you want to track it - it's not something I would necessarily rush to tackle unless other users are asking for it, due to the likely amount of effort involved.

Funkenjaeger, 2023-08-22:
> > It is off by default. Add `--include-tracks` and you probably also want `--include-nets` flags to your command.
> 
> Unfortunately adding the flags seems to lead to the same result.

Just to be clear, are you saying you're not seeing copper at all, or that the copper isn't correct (e.g. due to the limitation with cutouts in zones just mentioned)?

This is what I get with `--include-tracks` and `--include-nets` - note that all the other copper residing inside the large zones (copper pours) are present and can be highlighted by net as usual, despite the cutouts limitation:
[image]

If you're not seeing copper at all, are you sure that in addition to the flags, you also enabled display of Tracks and Zones in the options menu in the generated HTML file?

qu1ck, 2023-08-22:
> So, I think to address this I'd either have to do it in the parser (implement similar logic as Eagle does in its rendering, transform each zone into one or more much more complex polygons), or do it [...]

Doing this in javascript is not feasible, no canvas operations will possibly match EDA specific things like hatched zones, spike connections, corner rounding, netclass clearance rules and all other details that every package implements slightly differently.

I think it should be implemented in python or zones should be skipped all together, like in EasyEDA parser. Maybe proper zone support can be a differentiating feature of the eagle->generic json->ibom workflow if you are still supporting that.

> Unfortunately adding the flags seems to lead to the same result.

Check html settings, either reset them or enable tracks manually.

Funkenjaeger, 2023-08-22:
> Doing this in javascript is not feasible, no canvas operations will possibly match EDA specific things like hatched zones, spike connections, corner rounding, netclass clearance rules and all other details that every package implements slightly differently.

And for the same reasons I won't be able to get it perfectly right in the parser either.  My cursory thought was that if I were to attempt it at all, I'd only punch cutouts around traces/pads/vias naively using global clearance rules and not handle all the other, more nuanced details like those you mention - thinking that it would be 'close enough' for the general ibom use case of assembly/troubleshooting aid.

> I think it should be implemented in python or zones should be skipped all together, like in EasyEDA parser. Maybe proper zone support can be a differentiating feature of the eagle->generic json->ibom workflow if you are still supporting that.

I can drop zone processing from the parser.  
I can still support the genericjson workflow as long as Autodesk doesn't break it, like they did (and subsequently un-did) a while back.

qu1ck, 2023-08-22:
I would say "good enough" to be better than no zone support at all is having following:
* Having correct clearance to other nets (taking into account both global clearance and net to net clearance if specified)
* Having connections to items of same net, simple solid connection, ignoring spikes is fine
* Having any manual cutouts (i.e. keepout areas)
Not necessary:
* Spiked connections
* Hatching
* Corner rounding
* Min width and any other rarely used rules

I understand that even with these simplifications this is not a simple task and likely needs a 3rd party geometry library for all the polygon cutting/padding/fracturing operations.

I mentioned the ULP workflow in case it gives you access to the calculated zone polygon that is not stored in the board thus simplifying all the work. But I don't know if that is the case.

Funkenjaeger, 2023-08-22:
Copy that, makes sense.  I think my path forward is to remove zone processing for now and put in a wishlist/enhancement issue to capture the possible future addition of all that geometry processing in python.

> I mentioned the ULP workflow in case it gives you access to the calculated zone polygon that is not stored in the board thus simplifying all the work. But I don't know if that is the case.

It does, fortunately, and that workflow currently handles cutouts as a result.

jrzwee, 2023-08-22:
Here you can see the output I got in 2021 with the brd2json EAGLE ULP and the output I got with https://github.com/openscopeproject/InteractiveHtmlBom/pull/409 .

**2021 brd2json**
[image]

**2023**
[image]

qu1ck, 2023-08-22:
@jrzwee Yeah, that's what we were discussing above. In your second picture tracks are there, they are just covered by the zone because it doesn't have cutouts. If you care about correct rendering of zones use brd2json.

qu1ck, 2023-08-30:
Closing since the main issue was fixed. @Funkenjaeger you can open another issue for tracking disabling zones if you want or just send an MR when ready.

closed 2023-08-30 by qu1ck