openscopeproject/InteractiveHtmlBom #410: Feature Request: Double click on component in view to source; place

dimembermatt, 2023-08-23:
Thanks so much for the amazing application; I've generating htmlboms in all my projects as part of my SOP for a while now.

During the assembly process I've been going through the components individually and marking them off as sourced and/or placed; this usually involves scrolling on the PCB view to the component in question, edging my mouse back to BOM list, making sure I can get all the way to the sourced/placed column without losing where I am, and clicking the buttons. The mark when checked option really helps!

I'm wondering if it is feasible or desirable to have a feature where multi clicking the component will auto check the boxes and update the PCB marking. This could be as follows:

- a single click scrolls the BOM to the correct component and highlights the row (the current behavior)
- a double click marks the part as sourced,
- and a triple click marks the part as placed.

You could also have double clicks cycle between three states: (unsourced, unplaced; sourced, unplaced; sourced, placed).

Hopefully this wishlist feature is something useful for others as well as it would definitely speed up my current workflow and reduce the mental effort of making sure I'm updating the right component (instead of losing track and having to repeat the process to make sure).

Regardless, thanks so much for the team's work on this!

qu1ck, 2023-08-23:
I don't want to make any features that rely on what checkboxes are named or how many there are because they are meant to be user data. The only somewhat "special" checkbox is "placed" and it has it's own shortcut 'n' to step through.

Also it does not make sense to me to use the pcb rendering to mark the "sourced" checkbox. Sourcing is a process where you usually look at component list, either in bom or in your inventory system or on purchasing site. You don't typically look at the pcb and think "I want to buy the component that goes here". You do want to mark it as "placed" though so you can click it and press 'n' or click checkbox.

> edging my mouse back to BOM list, making sure I can get all the way to the sourced/placed column without losing where I am, and clicking the checkboxes

The 'n' shortcut should help here as well as dragging the checkbox column to the right, closer to rendering.

dimembermatt, 2023-08-24:
Thanks for the N shortcut tip, this is sufficient for me. Regarding sourcing, you are correct; I'm typically looking at my physical Mouser ordering listing while checking these off.

closed 2023-08-24 by dimembermatt