openscopeproject/InteractiveHtmlBom #411: COloring Nets

ankithjv, 2023-08-28:
When I color the nets can I Highlight all the pads as well and able to only color the pads connected to said net.

ankithjv, 2023-08-28:
Can I add zoom functionality as well when I click on something on the table for components. and Hover functionality when I hover over a pin on the layout that gives me pin number and part number.

qu1ck, 2023-08-28:
> When I color the nets can I highlight the complete net and all the pads as well and able to only color the pads connected to said net.

That is not supported currently. You should be able to tweak pad rendering color according to the net pretty easily with some custom js.

> Can I add zoom functionality as well when I click on something on the table for components. and Hover functionality when I hover over a pin on the layout that gives me pin number and part number.

These can also be done but 1) it should be in a separate wishlist issue 2) it's not well aligned with main use case of ibom and I will not prioritize it until templates are implemented.
If you want to implement it yourself, it's doable.

closed 2023-12-17 by qu1ck