openscopeproject/InteractiveHtmlBom #413: Strange Silkscreen Behavior

brettcoppolo, 2023-09-14:
**Issue:**
HTML view offsets any silkscreen that is not either a value or refdes. This issue may be macOS dependent, I have not tested on Windows. The issue appears to be present on multiple web browsers within macOS.

**Screenshots:**
Showing board close up: https://snipboard.io/oU4Dix.jpg
Zoomed out to show silkscreen offset: https://snipboard.io/jpY8ta.jpg
*Silkscreen offset can be seen in bottom right on board preview

**Version info:**
Version: 7.0.7, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.13.0
	HarfBuzz 7.3.0
	FontConfig 2.14.2
	libcurl/7.87.0 (SecureTransport) LibreSSL/2.8.3 zlib/1.2.11 nghttp2/1.41.0

Platform: macOS Big Sur Version 11.7.8 (Build 20G1351), 64 bit, Little endian, wxMac

Build Info:
	Date: Aug 14 2023 15:55:50
	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.82.0
	OCC: 7.7.0
	Curl: 7.77.0
	ngspice: 38
	Compiler: Clang 13.0.0 with C++ ABI 1002

qu1ck, 2023-09-15:
I can't reproduce.
Can you attach a board file that shows this issue? Feel free to remove everything that is sensitive and leave only board outline and the silkscreen elements that dont render correctly.

brettcoppolo, 2023-09-15:
Sure thing!

Here's the BOM as generated:
[InteractiveBOMTestPCB.html.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/12623432/InteractiveBOMTestPCB.html.zip)

Here's the board file that I generated the above BOM with:
[InteractiveBOMTestPCB.kicad_pcb.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/12623438/InteractiveBOMTestPCB.kicad_pcb.zip)

I hope that ZIP format is okay, I couldn't upload the files in their native format.

qu1ck, 2023-09-16:
Interesting. I see the issue in your generated bom file but I don't get the same result if I generate it from the board myself.

Few things I noticed: the offset seems consistent and scaled by same number. All text items also use a custom font.
Did you use some batch text modification function before generating the bom?
Do you get the offset silkscreen issue even after closing kicad, reopening the board and generating bom again?

In the past some kicad apis had issue where some update operations did not fully update the internal state correctly and the API was returning wrong values which were fixed after the board was reloaded. Maybe it's a similar case here?

qu1ck, 2023-09-28:
Can you share the font you use? I tried it with custom font on my end and can not reproduce it but it's not the same font.

brettcoppolo, 2023-09-28:
Sure! It's Microgramma D Bold Extended, and I think I downloaded the font from here: https://fontsgeek.com/fonts/MicrogrammaDBolExt-Regular

qu1ck, 2023-09-28:
Good news: I was able to reproduce it.
Bad news: it's a kicad bug https://gitlab.com/kicad/code/kicad/-/issues/15767

Most likely you will need to wait until it's fixed there unless I find a workaround.

qu1ck, 2023-09-29:
Turns out it was already caught and fixed for next build. You can get testing build from https://downloads.kicad.org/kicad/macos/explore/7.0-testing

qu1ck, 2023-10-09:
This is fixed in 7.0.8

closed 2023-10-09 by qu1ck