openscopeproject/InteractiveHtmlBom #415: Improve zone handling in Eagle/Fusion parser

Funkenjaeger, 2023-09-22:
As discussed in #408, zones are not currently handled entirely correctly in the Fusion/Eagle parser, particularly with regard to cutouts and spacing.  

In Eagle, the determination of those features is done internal to the software and that geometry is not stored to the .BRD file, so attempting to reflect them faithfully in ibom would require considerable effort.  That is a candidate for potential future enhancement, but in the meantime processing of zones in this case should be eliminated from the parser.

In Fusion, substantially more geometry information for cutouts/spacing in zones *is* saved to the .FBRD file and so there is the potential to represent zones faithfully.  Currently the zone parsing logic is too simplistic, and requires some additional work to fully and correctly handle zones.

Funkenjaeger, 2023-09-22:
Progress!  There are always gotchas, but I'm feeling optimistic about being able to make this work for the Fusion workflow at least.
[image]

Funkenjaeger, 2023-09-22:
@qu1ck have you given thought to adding a means for the holes in vias to be rendered?  I thought maybe I just hadn't figured out how, but I see that even in the kicad-based examples in the wiki the vias are filled:
[image]
Is it a case where because the drill size for a via is not reflective of its finished inside diameter (may not be an open via at all) it's better to not try to draw the hole at all?

This question is tangential to what I'm doing here, but it came to my attention again since sometimes zones have a lot of vias in them.

qu1ck, 2023-09-23:
I considered it when I was working on tracks support, I found that rendering via holes does not add to visual enough. Also in majority of cases vias are covered by soldermask and are not actual holes.

Funkenjaeger, 2023-09-23:
> Also in majority of cases vias are covered by soldermask and are not actual holes.

While it's certainly the norm for commercial boards to tent the vias (for good reasons), I for one don't tent vias on any of my boards, as I much prefer having them exposed for troubleshooting reasons (probing), so there would actually be some tangible benefit for my troubleshooting use case.
I think I'll go ahead and create a separate issue as a feature request for that, and we can continue discussion there.

qu1ck, 2023-09-24:
Thanks for the fix!

PR #417 (2023-09-24): Improve zone handling in Eagle/Fusion parser
Fix #415
closed 2023-09-24 by qu1ck