openscopeproject/InteractiveHtmlBom #416: Render holes in vias

Funkenjaeger, 2023-09-23:
It's typical, particularly in commercial boards, for vias to be tented (covered with soldermask) and therefore the current way ibom renders vias as solid zero-length tracks is appropriate:
[image]

However, for all my prototype/personal use boards I *don't* tent the vias, because exposed vias are a really convenient "free" test point for probing while troubleshooting a board, which is my secondary use case for ibom (after assembly).
Incidentally, un-tented vias like this are what I get using the default settings in Fusion/Eagle and fab at JLCPCB.
[image]

As such, it would be helpful in my use case to have my un-tented vias rendered with a hole in the center.  For vias on tracks, this would make them slightly more visually obvious, particularly for a via with an annular ring not significantly larger than the connected track.  Vias in zones are currently invisible, so this would make them visible - most commonly, this would be convenient for picking up ground when probing.

Additionally, it might also be nice if the vias are rendered in the same color scheme as through-hole pads (e.g. two-tone grayscale) so they're even more visible - which would seem logical, since an exposed via is in reality practically equivalent to a through-hole pad.

Funkenjaeger, 2023-09-23:
It probably goes without saying, but I would personally tackle implementing this for the Fusion/Eagle workflow, and possibly the genericjson workflow if desired.

As far as implementation/integration, my initial thoughts are:

- Obviously, vias still wouldn't be included at all if tracks aren't included
- Add an optional drill diameter parameter to tracks.  Would only apply when the track is zero length (i.e. is a via), and if this parameter isn't included then the via would be rendered filled, exactly the same as current behavior.  This way, I think this shouldn't affect kicad or other workflows (until/unless someone wants to add this behavior to those workflows also)
- I think that in Fusion/Eagle I can automatically determine whether any given via would be tented or not based on design rules in the file
- Simply rendering holes in vias seems innocuous enough in my mind to not warrant an additional command-line argument to enable/disable it... but I could add one if desired
- If I were to change via rendering to look like through-hole pads, that's a bit more significant of a visual change, so might warrant a command-line argument.  I'm open to suggestion on that, and whether the default should be enabled or disabled.

qu1ck, 2023-09-23:
I've checked kicad and there it's a board level setting. Either all vias are tented or none. Is that the case with fusion/eagle too? If so then there is no need for a making this optional, just render them like through hole pads if board settings call for untented vias.

> Add an optional drill diameter parameter to tracks. Would only apply when the track is zero length (i.e. is a via), and if this parameter isn't included then the via would be rendered filled, exactly the same as current behavior.

Sounds good.

Do you want me to implement format changes and kicad side first or do you want to do format + eagle first?

Funkenjaeger, 2023-09-23:
> I've checked kicad and there it's a board level setting. Either all vias are tented or none. Is that the case with fusion/eagle too? 

No, fusion/eagle has a design rule that's a (singular) threshold drill diameter.  Vias with a drill diameter smaller than the threshold get tented, larger are un-tented.  The default seems to be 0, thus all vias are un-tented by default unless the user updates the design rule.  Here's an example using the manufacturing preview in Fusion - I set the threshold at 14 mils (0.3556mm), and the three vias' drill sizes (from top to bottom) are 0.65mm, 0.35mm and 0.4mm respectively.
[image]

> there is no need for a making this optional, just render them like through hole pads if board settings call for untented vias.

Copy.  For fusion/eagle I'll do it on a per-via basis - seems super easy, I'll simply omit the drill diameter parameter from the track if it's below threshold, and that via will just render as tented.

> Do you want me to implement format changes and kicad side first or do you want to do format + eagle first?

I'll start a PR with at least the fusion/eagle changes and DATAFORMAT.md first, which should be quick.  I can then start poking at the JS (where I assume the change to render un-tented vias like through-hole pads will need to happen), although I'm a bit less savvy at JS so if you were to beat me to the punch on that part that's also fine :)

qu1ck, 2023-09-26:
Untented vias are now supported in kicad.

Heads up to @juulsA in case you want to add support in your Allegro export script. All you have to do is add `drillsize` attribute to track objects that are vias that you want rendered with a hole.

closed 2023-09-26 by qu1ck

juulsA, 2023-09-26:
I stumbled across this topic yesterday and I have already started implementing it ... support is coming soon!

juulsA, 2023-09-26:
> I stumbled across this topic yesterday and I have already started implementing it ... support is coming soon!

Done!

Funkenjaeger, 2023-09-26:
Well that escalated quickly!