openscopeproject/InteractiveHtmlBom #422: Make the top bar collapsible

mondalaci, 2023-10-22:
The top bar that contains the project name, revision, and various buttons occupies precious screen space. I seldom use the top bar features, and I'd much rather use this space for displaying my board, which is quite large.

Ideally, I want a collapse/expand button that hides/reveals the top bar.

It'd also be nice if the top bar were much more compact by default or configurable.

closed 2023-10-23 by qu1ck

mondalaci, 2023-10-24:
Thank you so much! Not only did you implement my request, you did it in record time! Works great.

Your tool is indispensable for creating new prototypes, and I heavily rely on it. Thanks for all your hard work!

qu1ck, 2023-10-24:
No problem. Yeah, stars aligned on this one: I had time, I liked the idea, I needed to cut a release this month and it fit well with a refactor that was well overdue. So I decided to just do it.

commit 7cf646f8 (2023-10-23): Add toggle to hide top bar

Fixes #422