openscopeproject/InteractiveHtmlBom #424: Feature request: Quiet / no-dialog mode

ckuhlmann, 2023-10-24:
I would like to suggest a parameter that suppresses all dialogue windows (and possibly drops the requirement for `wx`), e.g. for automated exports via terminal invocation.

I have a KiCad 7.08 PCB that uses a custom drawing sheet defined via `File>Page Settings>Drawing Sheet File` as `${TEST1}\test_sht.kicad_wks`. 
`TEST1` is defined in `KiCad's Main window > Preferences > configure Paths...` as the `sht` subfolder in my project directory by an absolute path (but could point anywhere else). The `sht` folder contains the `test_sht.kicad_wks` drawing sheet (see attached [test1.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/13121063/test1.zip) file).

If I open the PCB directly in KiCad, no warning or dialogue is shown and the drawing sheet is displayed.

If I export the PCB to an ibom via 
```
"C:\Program Files\KiCad\7.0\bin\python.exe" "C:\Users\xyz\Documents\KiCad\7.0\3rdparty\plugins\org_openscopeproject_InteractiveHtmlBom\generate_interactive_bom.py" "TestProj.kicad_pcb"
```
I get an error in the terminal "E r r o r   l o a d i n g   d r a w i n g   s h e e t .", but also a dialogue box 

[image]

Since I am trying to automate my exports using `kicad-cli` and would like to also automatically generate the ibom without user interaction, this is a bit of an issue, since the export stops.

I think that the dialogue comes from the function call in line 20 in `InteractiveHtmlBom/ecad/kicad.py`: `self.board = pcbnew.LoadBoard(self.file_name)  # type: pcbnew.BOARD` , so I'm not sure whether the ibom script has any control over this. If the dialogue could be suppressed, this would be great.

The test project that demonstrates the issue is included in the attached zip archive:
[test1.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/13121063/test1.zip)

Thanks for considering and for this very useful project.

qu1ck, 2023-10-24:
Yeah, that is coming from kicad and not the plugin, I don't have control over it.
You should report this to kicad bug tracker, using pcbnew bindings outside of kicad should not be throwing dialogs.

As a workaround, you should be able to define an environment variable (you can do it in a batch script) with same name/value as your kicad path variable, it should be picked up and the worksheet will be found.

ckuhlmann, 2023-10-25:
Thanks for the hint about the environment variables, it seems to work for the issue at hand.

As suggested, for a more general solution, I created [issue #15948](https://gitlab.com/kicad/code/kicad/-/issues/15948) in the KiCad bugtracker to request dialog suppression when running scripts outside KiCad or a quiet mode option for this use case.

closed 2023-10-25 by qu1ck