openscopeproject/InteractiveHtmlBom #427: Hello! Please tell me how to add this title. I want to add some properties to the components, such as the withstand voltage value of the capacitor and the error of the resistor, etc.

DaLaoHei1986, 2023-10-26:
[image]

qu1ck, 2023-10-27:
Assuming you are using kicad, edit the component fields in schematic, save it, update the pcb from schematic, save it, then use the "Fields" tab in plugin configuration to show/group the fields you want.

DaLaoHei1986, 2023-10-27:
> 假设您使用 kicad，编辑原理图中的组件字段，保存它，从原理图更新 PCB，保存它，然后使用插件配置中的“字段”选项卡来显示/分组您想要的字段。

Unfortunately, I am using Allegro. Do you currently have any tutorials in this area?

qu1ck, 2023-10-27:
According to JuulsA's comment in  https://github.com/juulsA/exportJson/issues/8 custom fields are supported by his export script.  In that case you just have to enable them in the bom config, either in gui if you use that, or with `--extra-fields` flag if you use only cli.

DaLaoHei1986, 2023-10-27:
> 根据 JuulsA 在 [juulsA/exportJson#8 ](https://github.com/juulsA/exportJson/issues/8) 中的评论，他的导出脚本支持自定义字段。 在这种情况下，您只需要在 bom 配置中启用它们，如果您使用它，则可以在 gui 中启用它们，或者使用 `--extra-fields` 标记（如果仅使用 CLI）。
Thank you very much for your reply. I'll try it now

closed 2023-12-17 by qu1ck