openscopeproject/InteractiveHtmlBom #428: Problem with parsing full arc board cutouts (allegro)

juulsA, 2023-10-29:
An issue has been reported to me where the board cutouts can't be parsed. At the moment I'm not really sure, if it's a problem with my script or if it could be a bug ... More specifically, it seems to be a problem when cutouts are made in form of a full arc. 

As you can see in the attached examples, the arc cutout is defined in line 42-49. If a full arc is defined by - in this case `"startangle": 360.0` and `"endangle": 0.0` several errors occur ([example_1.json](https://github.com/openscopeproject/InteractiveHtmlBom/files/13197815/example_1.json)). If you change the endangle to 359 degrees ([example_2.json](https://github.com/openscopeproject/InteractiveHtmlBom/files/13197816/example_2.json)`), the file can be fully parsed. Any idea what could be the mistake?

qu1ck, 2023-10-29:
Arcs are internally transformed into format compatible with svg, it requires distinct start and end points of the arc, otherwise it's shape is ambiguous.
For 360 degree arcs you should use Circle shape instead.

juulsA, 2023-10-29:
Thanks for your fast reply ... it seems that my query if a shape is a cirle is not working as expected ...

closed 2023-10-29 by juulsA