openscopeproject/InteractiveHtmlBom #429: EAGLE 9.6.2: Bottom side component incorrect orientation.

depalamarchuk, 2023-11-03:
Hello,

The script either rotates a certain component by 180 degrees or mirrors it (not sure what happens here). The component is located at bottom side. I'm using the latest version. I ran the command:
`python "Z:\CAD\Electric CAD Libraries\EagleCAD\01_Sys\ibom\InteractiveHtmlBom\generate_interactive_bom.py" "C:\ibom mwe\mwe.brd" --name-format "%f BOM Rev ibom" --show-fabrication --include-tracks`

This is from EAGLE 9.6.2:
[image]

And this is the HTML BOM:
[image]

I'm attaching a minimum project, so you can check it:
[ibom mwe.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/13248689/ibom.mwe.zip)

Thank you and have a nice day!

Funkenjaeger, 2023-11-04:
Thanks for sharing an example demonstrating the problem, that's always appreciated!
I'm away from my computer for about a week so it'll be about that long before I'm able to debug this.  

Interestingly, from your screenshots it looks like only the half-circle was inappropriately mirrored.  The line segments (2 on the left, 1 on the right) look correct.

qu1ck, 2023-11-04:
`L5` text is also rotated the other way.

Funkenjaeger, 2023-11-11:
Fixed U7 arc mirroring and L5 text rotation in #433.

[image]

commit c77c74a9 (2023-11-11): (eagle) Fix #429
closed 2023-11-11 by qu1ck