openscopeproject/InteractiveHtmlBom #434: CLI Execution for Eagle Board Files

LaCocoRoco, 2023-11-18:
Hi, I would love to use your project, but I have some issues and can't find any specific documentation. As per my understanding, the Run.bat is for EasyEDA files. Executing generate_interactive_bom.py via WSL stops with an import wx error. This is likely because it should be executed as a plugin in a KiCAD environment. In your repository's issue history, I have seen CLI usages of this kind:

```bash
generate_interactive_bom.py name.brd --name-format "%f BOM Rev ibom" --show-fabrication --include-tracks
```

However, this also results in an import wx execution error. WSL uses Python3. I also tried Python2 and installed some libraries to fix the import wx issue without success. Do you have any suggestions?

qu1ck, 2023-11-18:
If you have import wx error that means wxpython is not installed. How did you install it? You should use system wxpython package, not pip one on linux/wsl. If you use ubuntu it's called `python3-wxgtk4.0`.

If you are past import error and have some other error, show me your output. If it's complaining about not having a DISPLAY then read "Alternative Option" section [here](https://github.com/openscopeproject/InteractiveHtmlBom/wiki/Tips-and-Tricks#alternative-option).

Funkenjaeger, 2023-11-18:
Are you specifically wanting to run it in WSL, or do you just want to run it in Windows generally?

I run it in regular windows CLI, and all it took was installing wxPython via pip.  
I did try it in WSL just now, and after `apt install libgtk-3-dev && pip install wxpython` and using the `xvfb-run` utility from the page @qu1ck just linked to work around the X server thing, it worked.

LaCocoRoco, 2023-11-20:
Thank you @qu1ck and @Funkenjaeger, installing wxPython via pip in Windows worked like a charm!