openscopeproject/InteractiveHtmlBom #439: problem with HTML BOM plug-in, when I use it on the V7.99 version

yongyo, 2024-01-07:
# Description
I am using the V2.8 version of the HTML BOM plug-in, when I use it on the V7.99 version of kicad, the exported web version of the BOM file, there will be a lack of PCB tags, the text information on it is there, that is, the tags of the PCB components are not displayed, very strange, in the kicad7.0 version is good

I packed the source file in a zip file（22.zip）, please help me see what went wrong, I'd love to use this plugin on version 7.99
# Steps to reproduce

1. 
[image]
[image]

[image]

# KiCad Version

```plaintext
Application: KiCad PCB Editor x64 on x64

Version: 7.99.0-4189-gbe21c689e8, release build

Libraries:
	wxWidgets 3.2.4
	FreeType 2.12.1
	HarfBuzz 8.3.0
	FontConfig 2.14.2

Platform: Windows 10 (build 17763), 64-bit edition, 64 bit, Little endian, wxMSW

	wxWidgets: 3.2.4 (wchar_t,wx containers)
	Boost: 1.83.0
	OCC: 7.7.1
	Curl: 8.4.0-DEV
	ngspice: 42
	Compiler: Visual C++ 1936 without C++ ABI

Build settings:
```
[22.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/13852341/22.zip)

qu1ck, 2024-01-07:
Fixed. Get latest version of plugin from github master branch or wait until next release.

closed 2024-01-07 by qu1ck

yongyo, 2024-01-08:
Thank you so much

commit 700d65d1 (2024-01-07): Add PCB_FIELD drawing class, processed as text

Fixes #439