openscopeproject/InteractiveHtmlBom #442: Ibom is summarizing components with the same value, but different part number

fellpower, 2024-01-29:
i have 2 resistors, both 1206 - and 10k - BUT - different partnumbers (one is 5%, the other is 0,1%)

Both resistors get grouped in one group, but thats not right - and uncool if u want to place the components, because they are different resistors ^^

qu1ck, 2024-01-29:
You can configure how parts are grouped however you want, check the fields tab.
If you don't have part number in the fields list then you should add it as field in eeschema and update the board.

closed 2024-01-29 by qu1ck

fellpower, 2024-01-30:
What do u mean with "fields tab"?

i have 2 different resistors, with different partnumbers, but the same value, so why its getting grouped?

fellpower, 2024-01-31:
ahh, yeah, its easy eda - sorry. 

hmm, i found a workaround, but this would be nice in the ibom in the future maybe ^^

qu1ck, 2024-02-03:
It is now supported, get latest ibom from github master branch and pass `--show-dialog` to the launch command line. 
You will be able to configure additional fields.