openscopeproject/InteractiveHtmlBom #446: IBOM isn't rendering correctly on iOS devices

w9ran, 2024-04-25:
Hate to post this as an issue as I'm beyond pleased with this excellent tool - but I've looked through the issues and didn't find an answer:

I can't get the resulting ibom.html file to display properly on my iOS devices (iPhone 7+ and iPad Air).   The framework shows up but not the table or board images.   If it's supposed to work with Safari browser I'd appreciate knowing what configuration or other changes I should make.     Thanks for any suggestions and thanks again for this excellent tool.   Bob

qu1ck, 2024-04-25:
I don't have access to apple devices anymore and can not test this. But it works fine on desktop safari so even if I had the access I'm not sure I would put effort into fixing quirks of mobile safari. As a policy I only promise to support modern major browsers, mobile safari isn't one.

Most likely root cause is lack of support for some javascript feature and could have a simple fix. But unless someone investigates the issue and opens a PR I can't help much.

w9ran, 2024-04-25:
Fair enough, thanks for the reply.  Just thought it might be a setting or something on my end.

closed 2024-04-25 by w9ran