openscopeproject/InteractiveHtmlBom #447: Feature Request: Additional fields in the board image

gwideman, 2024-05-16:
It's great that IHB allows adding custom component fields to the BOM table, but I'm hoping to show a custom component field in the board image.  

IHB's board image already:
- allows turning on/off Fab and Silkscreen layers 
- allows turning on/off References and Values. 
- displays free-floating Texts that are on the Fab layer

I "just" need one footprint custom field that's on the Fab layer to display on the board image.

What are the prospects?

qu1ck, 2024-05-16:
If it's on the fab layer it should be displayed, doesn't matter if it's a custom field, a drawing or a footprint value. Just make the field visible in kicad and it will show up in ibom on fab layer.

qu1ck, 2024-05-16:
Hm, it seems to be broken in kicad 8.0.

qu1ck, 2024-05-16:
Fixed now but you'll have to wait for next release or install manually latest from github. Instructions for manual installation are on the wiki.

closed 2024-05-16 by qu1ck

gwideman, 2024-05-16:
OK! That's awesome!  What's the estimated date for next release?

qu1ck, 2024-05-17:
Probably some time next month.

commit a0887b33 (2024-05-16): Fix footprint fields handling for kicad 8

Fixes #447