openscopeproject/InteractiveHtmlBom #448: Doc for blacklist/whitelist?

gwideman, 2024-05-17:
Sorry if I didn't find the explanation, but...

What is the purpose and function of the blacklist/whitelist feature?

Thanks.

qu1ck, 2024-05-17:
The CLI help has some info (it's also on the wiki in usage page).

In short, that feature is used for board variants. You choose a field that contains variant name and the whitelist/blacklist will get populated with various values that your variant field has. You can then choose whitelisted variants (only components that have field set to one of the whitelisted values will be in the bom) or blacklisted variants (only components that don't have one of the blacklisted values will be in the bom).

You can also use it for other things like filtering out the tht/smd if you add a field for that and select it as variant.

gwideman, 2024-05-17:
OOOoooh... I get it now.  Somehow I didn't realize this was associated with the Variant feature (which I had not previously tried).

I now see that if I establish a particular field on components, and provide it with a value from some limited set of values, then this feature can filter those components.

OK, checkmark that value in the whitelist to include, checkmark blacklist to exclude.
Is there some particular logic when a value has a checkmark for both whitelist and blacklist. Or neither?

I see that filtering applies to the BOM table, but it does not seem to operate on the board image, where all components seem to appear regardless, is this right?

Here's our use case: we highly favor the image of the board showing only the footprints of components pertaining to a selected variant, and omitting any other components.

One particular problem where that clarity is important is when the components of alternative variants occupy the same space on the board. For example we have a number of boards that contain alternative power supplies (say for Mains input, or 12VDC etc) that should be populated with one or another set of components. 

The two (or more) sets of components occupy different sets of footprints, but those footprints all occupy the same board region, with one set of footprints overlapping the other set of footprints.  Part of the point of a variant-specific BOM and board image (which we currently make partly manually) is to make absolutely clear which footprints are to be populated, and which not. Graphics from the other overlapping components irrelevant to the particular build are a significant distraction.

Anyhow, hopefully this use case provides some input for future consideration. Thanks.

qu1ck, 2024-05-17:
> Is there some particular logic when a value has a checkmark for both whitelist and blacklist. Or neither?

Whitelist takes precedence. There is no reason to use whitelist and blacklist at the same time.

> I see that filtering applies to the BOM table, but it does not seem to operate on the board image, where all components seem to appear regardless, is this right?

Yes, that is intentional, like I mentioned in the other issue, board render is supposed to always show what the board actually looks like. Use "mark when checked" and "dnp outlined" features in html to affect rendering.

gwideman, 2024-05-18:
OK, thanks for your answer.

closed 2024-05-20 by qu1ck