openscopeproject/InteractiveHtmlBom #450: kicad: dnp has no effect on Fab layers

nopeppermint, 2024-05-21:
in my current pcb Project I have moved the component outline + designator text to F/B.Fab Layer.

If I now would like to only show the mounted components, this is only possible if the pads are displayed, as "dnp outlined" has no effect on Fab layer items...

I would suggest completely removing the Fab attributes in case "dnp outlined" is checked.
Or any other solution, which makes dnp option checked visible on Fab layers.

Would be nice, if that could be added, thanks

Example print screens:

with pads visible, everything is ok:
[image]
[image]

pads not  visible (disabled), no difference with "dnp outlined" active:
[image]
[image]

nopeppermint, 2024-05-21:
I am on commit https://github.com/openscopeproject/InteractiveHtmlBom/commit/a0887b33803ee912109b96742a8c2a1587c71d48

with Kicad:

```
Application: KiCad PCB Editor x64 on x64

Version: 8.0.2-124-g1264230973, release build

Libraries:
	wxWidgets 3.2.4
	FreeType 2.12.1
	HarfBuzz 8.3.0
	FontConfig 2.14.2
	libcurl/8.5.0-DEV Schannel zlib/1.3

Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW
OpenGL: ATI Technologies Inc., AMD Radeon(TM) 780M, 4.6.0 Compatibility Profile Context 22.40.70.16.231017

Build Info:
	Date: May 19 2024 05:22:18
	wxWidgets: 3.2.4 (wchar_t,wx containers)
	Boost: 1.83.0
	OCC: 7.8.1
	Curl: 8.5.0-DEV
	ngspice: 42
	Compiler: Visual C++ 1939 without C++ ABI

Build settings:

```

nopeppermint, 2024-05-21:
minimal example project to reproduce it:
[interactive_html_bom_example.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/15387080/interactive_html_bom_example.zip)

qu1ck, 2024-05-21:
This was discussed earlier, see https://github.com/openscopeproject/InteractiveHtmlBom/issues/336#issuecomment-1323722800

I'll close this as a duplicate, follow that issue for updates.

closed 2024-05-21 by qu1ck