openscopeproject/InteractiveHtmlBom #452: "Highlight first pin" option is cluttered in macOS

showier-drastic, 2024-05-25:
[image]

The three radio buttons are overlapped.

showier-drastic, 2024-05-25:
This seems to be a bug of wxWidgets (or wxpy) bundled by KiCad.

```python3
import wx

class Frame(wx.Frame):
    def __init__(self, title):
        wx.Frame.__init__(self, None, title=title, pos=(150,150), size=(350,200))
        panel = wx.Panel(self)
        box = wx.BoxSizer(wx.VERTICAL)

        highlightPin1Choices = [ u"None", u"All", u"Selected" ]
        self.highlightPin1 = wx.RadioBox(panel, wx.ID_ANY, u"Highlight first pin", wx.DefaultPosition, wx.DefaultSize, highlightPin1Choices, 3, wx.RA_SPECIFY_COLS )
        self.highlightPin1.SetSelection( 0 )
        box.Add( self.highlightPin1, 0, wx.ALL|wx.EXPAND, 5 )

        panel.SetSizer(box)
        panel.Layout()

app = wx.App()
top = Frame("a")
top.Show()
app.MainLoop()
```

Upstream wxWidgets and wxpy:

[image]

KiCad bundled wxpy:

[image]

qu1ck, 2024-05-25:
Thanks for checking if it reproduces on upstream, it does seem like something is wrong in kicad's fork of wx.

closed 2024-11-18 by qu1ck