openscopeproject/InteractiveHtmlBom #453: render.js function calls

ankithavn, 2024-06-14:
Hi, 

In the render.js file, where are the created functions called? Or are they referred to in a different file? 

Thanks!

qu1ck, 2024-06-14:
It's mainly called from `ibom.js` through `redrawCanvas()`

ankithavn, 2024-06-17:
Ohh okay, where is redrawCanvas() called in the self contained html file?

qu1ck, 2024-06-17:
Use the search function on github or in your IDE and you will easily see where it is called.

I'll give you brief overview of how it works but I can't teach you how to read code.
There is ibom.html template in the web directory, it has basic frame of the page and placeholders for javascript, css and pcb data. When python part generates the bom it compiles the pcb data into format that is described in `DATAFORMAT.md` and injects that data along with all the javascript and css into the template as output. The resulting html file has a `window.onload` handler which gets executed first thing as browser loads the page, that handler renders initial image and generates the bom table. Each time you resize the window or manipulate the bom in a way that affects the drawing things get rerendered again.

ankithavn, 2024-06-18:
Thank you! I seemed to miss the window.onload handler when initially reading the code and was getting lost. That cleared it up.

ankithavn, 2024-06-20:
I am using https://github.com/juulsA/exportJson in order to extracty json data from Allegro pcb files. And then I edited the json to add a field in every component, called "numdefects". When I try to generate the the html, I am led to an error that is thrown here in "genericjson.py". I cannot follow it further though and I was wondering where the validation occurs and if I can add this extra field in the expected json structure. 

[image]

qu1ck, 2024-06-20:
Validation happens here
https://github.com/openscopeproject/InteractiveHtmlBom/blob/master/InteractiveHtmlBom/ecad/genericjson.py#L41

Most likely you edited the json in a way that breaks schema compliance, make sure you read the schema https://github.com/openscopeproject/InteractiveHtmlBom/blob/master/InteractiveHtmlBom/ecad/schema/genericjsonpcbdata_v1.schema

The error message should tell you what the issue is.

ankithavn, 2024-07-11:
Thank you, that cleared it up.
 
How can I reference the components in the javascript as I am putting my defect data in the "extra_fields" section of components?
 
Also, in render.js, what type of object is pcbdata.footprints ? I thought it was a dictionary based off of the dataformat but then in "drawfootprints" and specifically line 388 I see that you iterated through it using pcbdata.footprints.length.

qu1ck, 2024-07-11:
It's an array, you can check such things in browser dev tools console.

ankithavn, 2024-07-12:
Okay, thanks! How can I reference components in the javascript? It is not a part of pcbdata upon checking devtools

qu1ck, 2024-07-12:
I don't understand the question, you reference items in an array as you normally would, by index or you just iterate over it with `for (.. of ..)` construct. What isn't part of pcbdata?

qu1ck, 2024-09-22:
Closing for inactivity.

closed 2024-09-22 by qu1ck