openscopeproject/InteractiveHtmlBom #455: big, comfortable checkboxes in table

mkdugi, 2024-07-12:
Hi,

First of all, thank You for your plugin! :-)

I'm trying to use in on cheap laptop with touchscreen and Linux, to help myself with soldering my plates.
But box to checkbox in "Sourced" and "Placed" columns are small and hard do hit with finger.
My proposal is to make this checkboxes as big as cells where they are placed. This cells exists only for this checkboxes, so area around checkboxes is not important.
This would make use of this plugin much easier on touching devices :-).

[image]

qu1ck, 2024-07-12:
I think keeping the default appearance but making the whole cell area act as checkbox on touch devices is better.

mkdugi, 2024-07-12:
> I think keeping the default appearance but making the whole cell area act as checkbox on touch devices is better.

Yes! It's absolutely very good solution! 
Unfortunately, i'm not a JS/HTML programmer :-(, I can't help. But I can test! :-).

commit 92b7a546 (2024-07-15): Use whole checkbox cell for touch events

Also fix keyboard shortcuts for checkboxes not working when they
are reordered relative to other columns.

Fixes #455
closed 2024-07-15 by qu1ck