openscopeproject/InteractiveHtmlBom #456: Inconsistent Handling of "PartNumber" Field in BOM Generation

anshulj07, 2024-07-17:
The goal is to extract various component details including "PartNumber", "Value", "Footprint", and custom fields to generate a comprehensive Bill of Materials (BOM).

Issue:
The "PartNumber" field, which is standard in our KiCad component definitions, is not being correctly displayed or included in the generated BOM. Unlike other default fields like "Value" and "Footprint".

Steps Taken:

I modified the generate_bom function to handle the "PartNumber" field. The function was adjusted to fetch and append the "PartNumber" from each component's data:

elif field == "PartNumber":
    fields.append(f.partnumber)

FYI: I am generating bom using command. 
    
However, it appears that there might be a mismatch or an error in fetching this field, possibly due to incorrect field naming or access method.

[image]

qu1ck, 2024-07-17:
`f.partnumber` is not a valid field of a footprint class, that's not how you access custom fields.

You should not have to modify the code at all, when you simply tick the part number in the ibom config, does it not work? Can you post an example board file where it does not work?

anshulj07, 2024-07-18:
This issue can be closed. 

So, I modified some code and moved the code to plugins. Now every time I export, I don't have to select part number. The reason I did this is because I some other use case where some API calls will be made to internal system to crawl over part numbers.

closed 2024-07-18 by anshulj07